logisim-evolution/logisim-evolution #1744: risc v assembler macro flaws

halvbanan, 2023-05-02:
i can't use % for macros, the assembler treats them for a immediate math operation and not for variable input, also throws me errors about parameter stuff and can't get inputs when calling the macro.
overall the risc v assembler is useful so far but i can't properly use the macros

BFH-ktt1, 2023-05-02:
Yes this is correct, the assembler and disassembler in logisim are very limited, it is just to be able to disassemble an elf-file generated with your favorite tools and being able to add just some assemble instructions inside. It is not previewed to put in a full-fetched assembler and disassembler in logisim, as this beat's it's purpose. Hope this explains.
Closing this issue as won't fix. Any PR's in this direction are of course very welcome.

closed 2023-05-02 by BFH-ktt1

halvbanan, 2023-05-18:
well it'd be useful to have at least la due to the limits of 12 bit immediate, if we can't implement it through macro then there is none