logisim-evolution/logisim-evolution #1747: RAM outputs Red when first loading LOGISIM

BlankedO, 2023-05-07:
After loading up LOGISIM, the output coming out of the RAM is red/error. It only happens every time you first load up the program, and it's fixed by enabling the RAM then ticking the clock once. I am not sure, but I think it is because when you first load up LOGISIM,  no value is coming out of the RAM, therefore the output wire is producing an error. So, any solution to this problem?
When I enable the RAM and tick clock it gets fixed. 

I tried switching Asynchronous read attribute to Yes as I have seen as a solution here, but it did not work.

[image]
[image]

maehne, 2023-05-08:
I can confirm your observation. I would expect the output being all `U`s instead of `E`s. Interestingly, it suffices to connect an input port to the address input of the RAM and then move the RAM component with the mouse to change the output value to all `U`s. In any case, you can only expect for a RAM to output something meaningful one clock cycle after applying the address. Interestingly, enabling the clear pin allows to force the output to all `0`s, but once clear becomes inactive the `E`s return.

langqingcui, 2023-05-12:
I want to raise the same issue, and seeing the red output really troubles me.

figurantpp, 2023-08-15:
Is there more work to do on this issue, or can it be considered as closed by #1824?

closed 2023-08-18 by maehne