logisim-evolution/logisim-evolution #1752: .deb claims mime type application/octet-stream

iehgit, 2023-05-19:
After installing logisim-evolution via the .deb package, all binary files are opened by default with logisim-evolution (and get its icon), because it gets registered as the application for the mime type application/octet-stream.
The postinst-script contains
`xdg-mime install /opt/logisim-evolution/lib/logisim-evolution-logisim-evolution-MimeInfo.xml`
I think that MimeInfo.xml is erroneous.
Contents:
`<?xml version="1.0" ?>
<mime-info xmlns="http://www.freedesktop.org/standards/shared-mime-info">
  <mime-type type="application/octet-stream">
    <comment>Logisim circuit file</comment>
    <glob pattern="*.circ"></glob>
  </mime-type>
</mime-info>
`
It not only claims files with the glob pattern *.circ, but everything with the mime-type application/octet-stream.
Please use a custom, specific mime type instead of application/octet-stream.

maehne, 2023-05-30:
Thanks @iehgit for the report! This is a duplicate of long standing issue #1471, which we have trouble to replicate on our side. Could you please check whether it helps to remove the `mime-type` line alltogether in `support\jpackage\linux\file.jpackage` and then building/installing the `.deb` file again?

maehne, 2023-07-07:
I am closing this issue as fixed since PR #1765 got merged. Feel free to reopen it if the problem should persist!

closed 2023-07-07 by maehne