logisim-evolution/logisim-evolution #1754: VHDL there is no change in the output.

wmurbina, 2023-06-02:
At the time of executing this entity in the output signal there is no change.
What could be the error?

LIBRARY ieee;
USE ieee.std_logic_1164.all;

ENTITY clktyp IS
    PORT (
        SIGNAL clk : OUT STD_LOGIC );
END clktyp;

ARCHITECTURE dataflow OF clktyp IS
BEGIN
    PROCESS
    BEGIN
    clk <= '1';
    WAIT FOR 2 ns;
    WHILE TRUE LOOP
        clk <= '0';
        WAIT FOR 5 ns;
        clk <= '1';
        WAIT FOR 5 ns;
    END LOOP;
END PROCESS;
END dataflow;

maehne, 2023-06-02:
Please provide a complete `.circ` demonstrating your issue. As we are not actively using the HDL co-simulation capability in our teaching, we are less familiar with that part of Logisim-evolution. Without a reproducer, it will be hard for us to provide support to you.

wmurbina, 2023-06-05:
Attached circuit.
[clktyp.zip](https://github.com/logisim-evolution/logisim-evolution/files/11652976/clktyp.zip)

maehne, 2023-06-07:
Thanks for providing the reproducer. The simulation engine of Logisim-evolution is event-based, but only has a more limited notion of time than the simulation engine of your VHDL simulator. Nevertheless, it acts as the master for driving the transient simulation of your circuit. At each point of time, where signals change at the inputs of the instantiated VHDL entity, it will let the VHDL simulator advance by a 100 ns time step to capture then the change of its outputs. This scheme is suited for zero-delay models, but not for models, where the VHDL entity generates signals in function of the advanced simulator time. You'll be able to observe changes on your VHDL entity's outputs once you add an input port and connect this to a changing (e.g., clock) signal.

I am closing this as "won't fix", as it works as expected. Feel free to reopen if you think the issue deserves further discussion. PRs to improve the simulation kernel or the co-simulation interface are welcome.

closed 2023-06-07 by maehne