logisim-evolution/logisim-evolution #1758: nios ii branch issue

halvbanan, 2023-06-11:
when branching to a label it jumps to the location but it negates the pc addition so it minuses when its ahead and plusses when its behind the branch

maehne, 2023-06-11:
Thanks for the report @halvbanan! Could you please provide a reproducer (ZIP archive with `.circ` file plus assembly/ELF file) demonstrating the issue?

halvbanan, 2023-06-12:
[branchtest.zip](https://github.com/logisim-evolution/logisim-evolution/files/11718708/branchtest.zip)
here, it just branches ahead of target and behind target, it should be a nop-branch but it executes a branch-branch

maehne, 2023-06-27:
Thanks for the reproducer @halvbanan! @BFH-ktt1: Could you please have a look, as you wrote that code?

BFH-ktt1, 2023-07-01:
@halvbanan : Thanks a lot for this bug report. I can reproduce your problem and will look into a sollution

BFH-ktt1, 2023-07-01:
@halvbanan : The problem should be fixed by my latest PR. Closing this issue as resolved, please reopen if there is still a problem.

closed 2023-07-01 by BFH-ktt1

halvbanan, 2023-07-03:
still is not fixed on the msi version, i redownloaded the latest version of x86-msi and ran the test that i gave on the zip folder, still does a branch-branch loop instead of branch-nop .

davidhutchens, 2023-07-04:
@halvbanan: You used the last release build. @BFH-ktt1 meant to say that the problem is fixed in our develop branch.

You can find the nightly build here:  [https://github.com/logisim-evolution/logisim-evolution/actions/workflows/nightly.yml](https://github.com/logisim-evolution/logisim-evolution/actions/workflows/nightly.yml)

For Windows, use the .msi artifact in the latest nightly build.