logisim-evolution/logisim-evolution #1761: Arduino C code

wmurbina, 2023-06-15:
Greetings.
Due to the difficulty for students to have an FPGA it would be possible to incorporate a module that allows to generate C code that compiles for Arduino.

maehne, 2023-06-16:
This is certainly not a priority for us and also problematic regarding the behavioural fidelity of such software-emulated hardware versus with respect to a real hardware implementation. The students risk to get a wrong impression on how real digital hardware behaves - especially when their designs do not consider correctly synchronous design principles, which is often the case. Then, they won't experience the effects of hazards, clock gating, asynchronous reset gating, bouncing, etc.

Also FPGA boards have become quite affordable in recent years. We should rather prioritize to improve the Verilog netlisting and support the open Yosys/nextPnR-based design flows for Lattice and Xilinx devices in addition to the current support for Altera/Xilinx toolchains.

I am closing this issue as not planned. Feel free to reopen if you would like to discuss this topic further. Also PRs in the direction proposed by you will be considered if they are general enough and don't impose too many limitations.

closed 2023-06-16 by maehne

wmurbina, 2025-03-21:
"Integration of Arduino Firmata with Logisim
I propose integrating the Firmata protocol into Logisim to allow digital circuits designed on the platform to be directly emulated on Arduino Uno and Mega boards, thus creating a bridge between virtual design and physical implementation.
Link: Firmata"

maehne, 2025-03-21:
@wmurbina: Feel free to submit a PR to propose that enhancement. However, my [comment above](https://github.com/logisim-evolution/logisim-evolution/issues/1761#issuecomment-1594455139) still holds. Code generation for Arduino should be handled through separate dialogues than download to FPGA boards, e.g., accessible via entries in the Project menu. The documentation for that new feature should make clear that the level of fidelity to an actual implementation using digital gates is limited.