logisim-evolution/logisim-evolution #1764: Add support for 7-segment scanning arrays

BFH-ktt1, 2023-06-20:
Many FPGA-boards have arrays of 7-segment displays, where the segments are connected together, and the display is selected by either a display enable pin (common anode or common cathode pin) like in the Alchitry (thanks @chuckb for the great video), or a decoder like on the GECKO5Education. By adding this IO-component, logisim can map a/multiple standard 7-segment display on one or multiple displays.

BFH-ktt1, 2023-07-27:
Closing this issue as #1786 got merged

closed 2023-07-27 by BFH-ktt1

JayTaq, 2025-09-07:
How to use this 7-segment scanning arrays in Logisim Evolution schematic?

maehne, 2025-09-08:
You don't use 7-segment scanning arrays directly in the schematic. Instead, you just use normal 7-segment displays (or even LEDs). The scanning array logic is added in the background by the FPGA Commander while net listing your design.  The properties of a scanning 7-segment array have to be defined only once in the FPGA board editor for each board featuring such displays. You can, e.g., open the built-in board `ALCHITRY_AU_IO` in FPGA > FPGA Board Editor and then double click on the upper left big rectangle to open the properties of the scanning array of that board:

[image]

[image]