logisim-evolution/logisim-evolution #1771: Add dot matrix into sub-circuit appeareance

nake90, 2023-07-09:
Hello! And first of all, thanks for this amazing tool!

tldr: As the title says, I would like to request adding the sub-circuit view of dot matrices.

I am heavily using the sub-circuit feature, but I am having trouble while developing a screen-like circuit using led dot matrices because they do not seem to have a viewable state in the circuit appearance window.

Some IO components like leds and 7segs do have it, but the dot matrix does not, which is making my life really hard.

If someone could implement that, that'd be awesome, but if you are all too busy with other features I'm happy trying to develop it myself. I have no idea about java programming, but I am competent in C++ so I started glancing over the code here:

```
src/main/java/com/cburch/logisim/std/io/
```

Components that show in the circuit appearance window seem to have a *Shape.java file that defines how to draw it, so I created a placeholder one for the led matrix using the led one as a basis.

Then I also noticed that each shape is being registered as an svg element, so I added my own `visible-matrix` entry to my DotMatrixShape.java and into `AppearanceSvgReader.java`.

Am I in the right path with this? Any pointers on how to continue/what to be careful about? I guess there must be a way to register my shape into the "viewable state" dropdown, but I am still not sure how to do that and if I am missing something else too. Any help is really appreciated.

maehne, 2023-08-28:
@BFH-ktt1 is familiar with this aspect of the code base and should be able to confirm whether your approach is right. Your enhancement proposal sounds anyway useful to me.

maehne, 2026-08-21:
Closing this as resolved by PR #2384.

closed 2026-08-21 by maehne