logisim-evolution/logisim-evolution #1780: Problem with the appearance of a sub-circuit V3.9

mbaillif, 2023-07-21:
The drag and drop of a sub-circuit does not take on a customized appearance. It always looks like the logisim.

maehne, 2023-08-28:
I agree with you that the current behaviour is non-intuitive. Also others raise regularly a lot of [issues regarding the custom appearance](https://github.com/logisim-evolution/logisim-evolution/issues?q=is%3Aissue+appearance). The current situation is therefore unsatisfactory. 

IMHO, the appearance should automatically set to custom once you edit the circuit's appearance. Personally, I would like the starting point for the editing be based on the appearance set right before the editing. From that on, the sub-circuit should consistently use this custom appearance until one removes/resets *explicitly* the custom appearance. Only then, it should be possible again to switch between the classic, evolution, and holy cross default appearances.

What do you think?

maehne, 2023-08-28:
See also issue #1280.

mbaillif, 2023-09-10:
Indeed, it sounds just as you describe it.

maehne, 2025-06-20:
I am upgrading the priority to standard, as we are frequently getting duplicate reports about this unintuitive behaviour. It would be nice if someone could look into how this usability issue could be resolved.