logisim-evolution/logisim-evolution #1782: N-type transistor

a4ch, 2023-07-21:
[image]

output of transistor should be activated

davidhutchens, 2023-07-21:
No. A transistor is not activated if its input (source) and control line (gate) have the same value. Look at the full table in the documentation and ignore the incorrect "summarized form", which we should remove. See #1295 for more discussion of why it works the way it does.

closed 2023-07-21 by maehne

a4ch, 2023-07-21:
> <logisim-evolution/logisim-evolution/issue/1782/issue_event/9885319133@
> github.com>

https://youtu.be/CPp3Fp3Y2dw

4:15

a4ch, 2023-07-21:
https://youtu.be/CPp3Fp3Y2dw

 4:15

davidhutchens, 2023-07-21:
@a4ch: We are aware that the original Logisim gave a 1 output for the case you gave above. However, that is not how the transistor actually works in hardware. The behavior was changed in Logisim-evolution years ago. Again, see #1295 for more.

maehne, 2023-07-21:
To complement @davidhutchens: The definition of the output behaviour of the transistor components in Logisim-evolution ensures that common digital logic topologies ([NMOS](https://en.wikipedia.org/wiki/NMOS_logic), [PMOS](https://en.wikipedia.org/wiki/PMOS_logic), [CMOS](https://en.wikipedia.org/wiki/CMOS), [transmission gates](https://en.wikipedia.org/wiki/Transmission_gate)) can be modelled. When creating MOSFET circuits in Logisim-evolution, pay attention to the following rules:

* PMOS transistors conduct well '1'. They are used to construct a current path from Vdd to the output (to load its capacitive load).
* NMOS transistors conduct well '0'. They are used to construct a current path from GND to the output (to discharge its capacitive load).
* The small arrow in the transistor symbol points towards the output.

Here's a small summary (see also [mosfet_topologies.zip](https://github.com/logisim-evolution/logisim-evolution/files/12134794/mosfet_topologies.zip)) what works and what not:

[image]

*Note:* If you just use a single NMOS or PMOS as pass transistor, you need to connect a pull-up or pull-down resistor to the output to get a well-defined logic level in all cases. Transmission gates contain a NMOS and PMOS transistor and thus conduct equally well '0' and '1'. This is exploited in [pass gate logic](https://redirect.cs.umbc.edu/~cpatel2/links/640/lectures/lect16_combo3.pdf).

R3dst0ne, 2023-07-24:
@davidhutchens & @maehne
Since this is a common issue, I created [a section regarding this](https://github.com/logisim-evolution/logisim-evolution/wiki/Logisim-evolution-FAQ#transistor-behavior) in the Wiki some time ago.
In its current form, it is better than nothing, but not as good as your explanations in the various issues.
Do you think it is possible/feasible to write one good explanation, that than can be linked in all issues regarding this topic?