logisim-evolution/logisim-evolution #1783: Fix transistor documentation

davidhutchens, 2023-07-21:
Our current transistor documentation does not accurately describe its behavior. The full table of behavior and the "summarized form" do not describe the same behavior. The wording describing the East edge output is also inaccurate.

mbaillif, 2023-07-24:
Hello,

In parallel to Logsim-evolution, I'm in the process of modifying and adapting the documentation to version 3.9.

The transistor parts and the memory group are already there, and if you'd like to review them, I'd be delighted to do so. 

I'd appreciate it if someone would proofread all the pages with a _V in the document cell docstatus.html and libstatus.html but obviously we can stick to the transistor page. Click on the table cells in docstatus.html or libstatus.html to open the document(s).

github branch: mbaillif/Logisim-evolution-documentation
It's available at: https://mbaillif.github.io/Logisim-evolution-documentation/
or in  www.baillifard.com/logisim

There are still a lot of mistakes to correct, but I'm making progress. That's over 100 hours of work.

What we can do:
Minimal: Push Transistor documents into the logisim-evolution repository.

Better: Push Transistor and memory documents into the logisim-evolution repository.

But in the end, it would be nice to push all the Logisim-evolution-documentation documents into the logisim-evolution repository.  But I'd like someone to go through the documentation first.

R3dst0ne, 2023-07-24:
A while ago, the idea came up to move the documentation to its own repository, because it would make editing a lot easier and enable us to do things like @mbaillif did. 

Now that LSe has its own organization, it is IMO time to do this step.
Since the doc source will still be needed in this repo, the newly created repository can be included as a [git submodule](https://git-scm.com/book/en/v2/Git-Tools-Submodules).

What are your opinions on this?