logisim-evolution/logisim-evolution #1801: Am I allowed to create a Hotkey Settings TAB in the pref?

gtxzsxxk, 2023-07-27:
Dear maintainers,
Different people have different using habits. Such as the undo. Many people will set them as "Ctrl+Y" rather than "Ctrl+Shift+Z". So I spend some time creating a new tab in the preferences, allowing users to change the hotkeys to whatever they like.

My repo and changes can be viewed here: https://github.com/gtxzsxxk/logisim-evolution
Looking forward to your acceptance.

gtxzsxxk, 2023-07-27:
I have uploaded the `changes.md`. My changes are as follows.
* @gtxzsxxk (2023-07-27)
  * Added a new class HotkeyOptions, a new TAB in the preferences.
  * Modified Menu class, setting up a new abstract method so that the hotkey can be sync via the gui
  * Modified MenuFile MenuEdit MenuSimulate to implement that abstract method
  * After my modification, 11 menuitem in the menu-bar are now available in hotkey settings. Changes will be updated immediately.
The 11 menuitem are as follows.
    * ###### Simulation
    * Auto-Propagate
    * Reset Simulation
    * Single-Step Propagation
    * Manual Tick Half Cycle
    * Manual Tick Full Cycle
    * Auto-Tick Enabled
    * ###### Edit
    * Undo
    * Redo
    * ###### File
    * Export
    * Print
    * Quit
  * I'd like to insert more hotkeys to make logisim-evolution more fun. Such as press *SPACE* to rotate the component to be placed on.

maehne, 2023-07-27:
@gtxzsxxk: Refarding rotation using the SPACE key: You are aware that you can directly orient components using the arrow keys? What’s missing for some components is a handy way to flip/mirror them: The splitters and transistors come to my mind: For this kind of task, the SPACE key might be an option. What do you think?

gtxzsxxk, 2023-07-27:
@maehne I totally agree with you. Let me try it tomorrow

gtxzsxxk, 2023-07-30:
> The splitters and transistors come to my mind: For this kind of task, the SPACE key might be an option. What do you think?

@maehne I find this quite complicated. Take splitter for example, to mirror a splitter, we have two ways.

One way is the most simple way. Note that a splitter has two attributes. They are Facing and Appearance respectively. Saying that we want to mirror a east-facing splitter, we just need to set Facing to west and Appearance to Right-Handed. Then it would look like being reflected along the Y axis.

The other way is to directly rewrite the component's draw method.

If we want to implement the mirror operation just for the splitter, it is quite simple using the first way. However, if we want to implement the mirror operation for all components, it will seem quite complicated. Because I find each component has its own drawing code and not supports the simple combo mirror way which can be used in splitters, i.e. if we want to implement the mirror operation for every component, we have to use the second way, i.e. we need to rewrite its drawing code., which is nearly impossible.

So I am puzzled out. If we only implement the operation for the splitter, it is OK but seems unnecessary because it doesn't make sense to set up a new hotkey binding for just one simple component. But to implement the operation for each component, it is hard. I don't know what to do now. Could you please give me some hints?

davidhutchens, 2023-08-16:
@gtxzsxxk: Is this issue now complete with the merger of PR #1802? If so, please close. If not, what remains?

gtxzsxxk, 2023-08-17:
Oh, sure, let me close that. Thanks!

closed 2023-08-17 by gtxzsxxk