logisim-evolution/logisim-evolution #1806: Problem when editing viewed circuit's appearance in VHDL code mode

gtxzsxxk, 2023-07-29:
Sorry to disturbing you again and again. There occurs a NullPointerException when I am clicking the small icon named "edit viewed circuit's appearance" when I am writing some VHDL. This is the console output.
```
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException: Cannot invoke "com.cburch.logisim.circuit.Circuit.getWidthIncompatibilityData()" because the return value of "com.cburch.logisim.proj.Project.getCurrentCircuit()" is null
	at com.cburch.logisim.gui.main.CanvasPainter.drawWidthIncompatibilityData(CanvasPainter.java:53)
	at com.cburch.logisim.gui.main.CanvasPainter.paintContents(CanvasPainter.java:201)
	at com.cburch.logisim.gui.main.Canvas.paintComponent(Canvas.java:524)
```

I simply troubleshooted and found that the problem lies in CanvasPainter.java line 104
```
private void drawWithUserState(Graphics base, Graphics g, Project proj) {
    final var circ = proj.getCurrentCircuit();
    final var sel = proj.getSelection();
   
    /* codes */

    final var context = new ComponentDrawContext(canvas, circ, circState, base, g, false);
    context.setHighlightedWires(highlightedWires);
    circ.draw(context, hidden);
    sel.draw(context, hidden);
```
The object `circ` is `null` at the beginning, then `circ.draw(context, hidden)` will throw a exception. I find a simple solution. It is to add a if condition to determine whether the `circ` is null. If so, directly return. This works.

But I am 100% not familiar with the code here, I don't know what I am doing.... So I just report this problem....
Wish you a good day.

maehne, 2023-08-25:
@BFH-ktt1: Could you maybe have a look, as you are familiar with this part of the code base?

BFH-ktt1, 2023-08-25:
@maehne : I would love to, also to improve the speed of the simulator as reported in #1805 by @lorenzonotaro and investigated by @davidhutchens, but unfortunately due to my two master courses at EPFL and my sabatical at BFH, I will probably only have time to release a new version before semester starts. Any help is appreciated :-)

maehne, 2025-09-05:
@BFH-ktt1: Do you think you could have a look into this issue before the upcoming 4.0.0 release? Otherwise, we can move it to version 4.1.0.