logisim-evolution/logisim-evolution #1807: Add hot keys for Project menu's Move Up & Move Down options

prlaba, 2023-07-31:
Please consider adding hot keys for the Project menu's _Move Up_ and _Move Down_ options.

It's tedious to have to reopen the menu and click on the option to move a circuit up or down one position, especially if the circuit has to be moved up or down many positions.

gtxzsxxk, 2023-07-31:
I'm working on the hotkeys these days: #1802 . If merged, users can edit the hotkeys in a preference Tab. But in my PR, I haven't consider the default key bindings for _Move up_ and _Move Down_. Do you have any candidates for them?

prlaba, 2023-08-01:
Hmm…. I hadn’t thought about specific hotkeys for those options.

How about Ctrl+U for Move Circuit Up and Ctrl+D for Move Circuit Down?  Are they already in use?

Paul

prlaba, 2023-08-03:
> @prlaba The combo Ctrl+U is very lucky, which is unused currently. But the second key binding Ctrl+D has been used as duplicate function in the Edit Menu. What about using Shift+U and Shift+D instead, since the shift key has the meaning of moving?

Yes, after I sent those suggestions I saw that Ctrl+D was already assigned.

Sure, Shift+U and Shift+D would work.

You’re planning to make the hotkeys editable?  That’s great!

Thanks,

Paul

davidhutchens, 2023-08-17:
@prlaba: With the merger of #1802, shortcuts for these actions should be implemented with ctrl-shift-U and ctrl-shift-D in the main branch. Please check our Nightly build, or build yourself from main.

I am closing this as completed. Please reopen if the issue is not solved.

closed 2023-08-17 by davidhutchens

prlaba, 2023-08-17:
Thank you!

Paul