logisim-evolution/logisim-evolution #1808: Suggestion: Please reduce size and change color of POR component

prlaba, 2023-07-31:
Please consider reducing the size and change the color of the POR component.  The large orange rectangle is out of character with all of the other components.  It looks cartoonish and takes up too much space in a circuit.

A small white square with a single output and "POR" in its center would be more in tune with the style of other components.

mbaillif, 2023-09-01:
I launch a PR to adapt this component

davidhutchens, 2023-09-10:
This issue should be fixed in the main branch with #1843.

@prlaba: would you please check that and either let us know what else should be done or close this issue as completed.

prlaba, 2023-09-11:
I’d be happy to verify, but don’t know how to build/download/install on Windows from main branch .  Instructions somewhere?

Thanks.

maehne, 2023-09-11:
> I’d be happy to verify, but don’t know how to build/download/install on Windows from main branch .  Instructions somewhere? Thanks.

You can check out our [nightly builds](https://github.com/logisim-evolution/logisim-evolution#nightly-builds).

mbaillif, 2023-09-12:
I also have a little help here for a Windows installation.
https://www.baillifard.com/logisim/en/html/guide/install/index.html

it's still only a draft

maehne, 2024-08-16:
Closing as fixed by PR #1843, which is part of Logisim-evolution 3.9.0.

closed 2024-08-16 by maehne