logisim-evolution/logisim-evolution #1809: Two loaded libraries can't share components from a third library

prlaba, 2023-07-31:
This bug is a bit complicated to explain.

Suppose I have three libraries I created, LibA.circ and LibB.circ.  LibA and LibB are standard circuits, while LibC contains some custom components.  LibA and LibB each use some of the components in LibC.

Now suppose I want to create a master circuit that uses circuits from the LibA and LibB libraries.

When I load the LibA library, it loads with no problem.  But when I attempt to load the LibB library, I get an error message complaining about any LibC components shared between LibA and LibC.

Apparently Logisim-Evolution doesn't allow adding two _user_ libraries that share components from a third user library.

The same thing happens if I load LibB first, followed by LibA.

Note that L-E _doesn't_ complain if the LibA and LibB libraries share components from L-E's built-in libraries.

But I can't figure out why this should be a problem.  The shared LibC components are the same in LibA and LibB, so there's only one set of LibC components.  So what's the problem?

My workaround was painful.  I had to create empty circuits in my master circuit that matched every circuit in my LibA and LibB libraries, then copy/paste each of those circuits from LibA and LibB into the corresponding circuit in the master circuit.  Now everything worked as expected; the new circuits could use components from LibC without any problem.

Of course, this has one significant side effect: the master circuit file now contained all of the circuits from the LibA and LibB libraries, so I could no longer manage the LibA and LibB circuits as their own files.  Not how I wanted to manage my circuits, especially ones with many subcircuits.

I don't think what I'm trying to do here is unusual, especially with complex projects that contain multiple levels of circuits, each with their own components and subcircuits.

maehne, 2023-08-25:
I can confirm the issue, as it was observed by our students in the past during their Logisim projects, in which they collaborated in small groups and tried to better organise components for reuse. A fix for this issue would be highly appreciated. Unfortunately, we have come up only with workarounds, e.g., the merge feature implemented by @BFH-ktt1.

prlaba, 2023-08-25:
I discovered this was my own error, not L-E’s. My two circuit files that each loaded a common circuit file as a library were in fact loading two different circuit files (the same file but in two different folders). Once I loaded the same circuit file into both files, all was good.

Nonetheless, L-E’s rules can be exasperating at times. For example, you can’t name a circuit file “Memory” because it conflicts with L-E’s  built-in Memory group. And you can’t load load two circuit files that include a same-named  circuit.

It’s too bad someone hasn’t written a complete Logisim-Evolution User Guide that covers this sort of stuff.

Paul

maehne, 2023-08-28:
Using libraries remains fragile, despite it is working now in your case. It's up to you to decide whether you want to close this issue.

> It’s too bad someone hasn’t written a complete Logisim-Evolution User Guide that covers this sort of stuff.

We are always looking for help to improve the documentation! PRs are welcome. Unfortunately, our resources are limited. @mbaillif is our most active contributor to the documentation. Nevertheless, we need to make it easier to contribute to the documentation (see, e.g., issues #303 and #824).

maehne, 2026-09-07:
Thanks @hewzhew, I agree and close this issue as resolved. @prlaba: Please reopen/comment if this issue persists with the current nightly builds.

closed 2026-09-07 by maehne