logisim-evolution/logisim-evolution #1814: Lock the circuit

ainoue2019, 2023-08-05:
I have used the logisim evolution for several years to educate students in our university.
When I give an assignment to students with logisim file, some students accidentally change the circuits.
I want some features to lock the circuits to avoid students to modify the circuits.
I appreciate if you could add this feature in the future.

Best Regards,
Atsuki

BFH-ktt1, 2023-08-06:
Dear @ainoue2019 : First of all thank you for using logisim-evolution and your feature request. To understand your "lock" feature a bit better:
1) Do you want to be able to lock single gates, wires, etc. or:
2) Do you want to be able to make certain circuits in your design "read-only".
In case 2), logisim already provides this feature by binding a .circ-file as a library. circ-files used as libraries cannot be changed (hence are read-only) in the current working set. Of course a student can open the circ file separately and modify it, but this is an extra step.
In case 1) this would be an extra attribute to all the components that you would have to set for each and every instance, which might be a lot of work (each wire-segment would have to be changed to the lock state). Of course, also here it would be easy for the user to "unlock" and edit.

If your needs are satisfied wit option 2) above, this would be great.
For option 1) this would be some significant work, for which I do not have the time to implement, but I'm happy to put it on the wish-list.

ainoue2019, 2023-08-06:
Dear Professor,

 

Thanks for your response.

 

I want to lock each circuit, not single gates.

I understand the feature of library has already enabled to lock the circuits(read only mode).

I will try it.

Thanks again for this useful information and please close my request on the github.

----------------------------------------------------

Atsuki Inoue

BFH-ktt1, 2023-08-06:
Closed as requested

closed 2023-08-06 by BFH-ktt1