logisim-evolution/logisim-evolution #1822: Recording anomaly in timing diagram.

mbaillif, 2023-08-11:
To test: I have this very simple circuit (for demonstration purposes).
[image]

Initial setting : 
  Continuous real.time mode, 
  Time scale 1,0s, 
  Capture fine-grained details Off or On
  After resetting the simulation, I have 0 input and output.

Step : 
  Reset simulation Ctrl+R, wait 3s, click on input, wait 1s, click on input, wait 2s, clock on input, Stop simulation Ctrl-E

I'm expecting this diagram ____3____/‾1‾‾\___2__/‾
but in fact /‾‾‾‾3‾‾‾‾\_1_/‾‾‾2‾‾\_
 
It's as if the state at the beginning was not taken into account and replaced by the state after the first click, but the times are correct, just shifted with respect to the signal value.

Another anomaly: if you set very small values in Time scale (< 10ns), the initialization part is recorded on the graph over a very long period.

mbaillif, 2023-08-11:
[image]

curves a little better :-)
Top: what you'd expect.
Bottom: what is.

maehne, 2023-08-28:
I can reproduce the issue.

maehne, 2023-08-28:
I observe the same issue in @kevinawalsh's Logisim-evolution Holy Cross edition 5.0.2.