logisim-evolution/logisim-evolution #1823: Another anomaly in the Timing Diagram

mbaillif, 2023-08-13:
To demonstrate, I use this very simple circuit:
And Stop-motion mode has Time scale: 5.0us
[image]
In the simulation we obtain this correct diagram.
[image]

However, if you modify the circuit or add or remove a signal, there's a problem later on.

**To demonstrate** :
- Build this circuit (above)
- Start timing diagram
- Simulation reset
- A few clicks on **s1** to advance the simulation  
- click "Add or Remove Signals" button and remove the **o2** signal,  
- Add it again (**o2**)
- A few clicks on **s1** to advance the simulation 
- Note that the **o2** signal appears at the point where it is added to the diagram.
[image]
- Reset de la simulation
- A few clicks on **s1** to advance the simulation 
- The O2 signal does not appear at the start of the frame, but later, and is out of sync.
[image]

maehne, 2023-08-28:
I can confirm the behaviour. To reproduce it, it is important to delete the ouput port **o2** and not just the wire connecting to it! It's really annoying, as a simulation reset is not fixing the issue.

maehne, 2023-08-28:
Also this bug can be reproduced in @kevinawalsh's Logisim-evolution Holy Cross edition 5.0.2.

maehne, 2023-08-28:
Interestingly, the problem can be worked around to some degree by removing/re-adding all signal using the "Add or Remove Signals" button, but then all traces start with an offset.