logisim-evolution/logisim-evolution #1827: error with ./gradlew  build

SetOfAllSets, 2023-08-16:
```
> Task :checkstyleMain FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':checkstyleMain'.
> Expected file collection to contain exactly one file, however, it contains no files.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 14s
13 actionable tasks: 13 executed
```

Gentoo linux

davidhutchens, 2023-08-16:
Work on the codebase style is incomplete. There are parts that fail checkstyle.

Try `./gradlew build -x checkstyleMain -x checkstyleTest`

figurantpp, 2023-08-19:
Are pull requests intended to fix style issues in the java code welcome? Or should we fix those at a later time to avoid merge-conflicts with existing pull requests?

davidhutchens, 2023-08-19:
Pull requests that fix style issues are welcome. We do ask that they be limited in size so they are easier to review and if they happen to introduce a bug, it will be easier to figure out what change caused the problem.

We are trying to move toward [Google Java Style Guide](https://google.github.io/styleguide/javaguide.html).

davidhutchens, 2023-08-23:
@SetOfAllSets: Are you using the current main branch? A similar problem was fixed in #1624. If you are using the master branch, you may not have that fix.

davidhutchens, 2023-09-10:
Since @SetOfAllSets has not responded to my question, I am assuming the issue is fixed in the main branch and I will close this issue. If the problem persists, please reopen this issue.

closed 2023-09-10 by davidhutchens