logisim-evolution/logisim-evolution #1828: 0-A=6?

Genious28, 2023-08-16:
I'm trying to use 2x 7-Segment Displays to Display 1 Hex Digit, in order to achieve that without complicated truth tables and stuff, i simply Subtracted the Counter/Clock by A (10) to get the Right Most Segment to Display 0-5 after 9 insead of A, B, C, D, E, F.. (Also because i am using a BCD to 7-Segment Component)
[image]

Unfortunately, due to some reason.. after 15 (F), the Right Most Segment Continues counting from 6-9 because the Subtrator apparently passes through the Output even if the Subtraction cannot be done.
[image]
[image]

I'm not sure if this is intentional or not, nor did i check if the Original Logism has this issue.. Because, well, the original Logism doesn't get Updates anymore.

Genious28, 2023-08-16:
This is the resulting Output due the the Error from the Subtractor;
(First number is the Data Line, the Second 2 Digits is the Output Segments Respectively)

0. 0**6**                                         0-A=6?  (Negative Value)
1. 0**7**                                         1-A=7?  (Negative Value)
2. 0**8**                                         2-A=8?  (Negative Value)
3. 0**9**                                         3-A=9?  (Negative Value)
4. 00                                         4-A<0    (Negative Value)
5. 00                                         5-A<0    (Negative Value)
6. 00                                         6-A<0    (Negative Value)
7. 00                                         7-A<0    (Negative Value)
8. 00                                         8-A<0    (Negative Value)
9. 00                                         9-A<0    (Negative Value)
10. 1**0**                                       A-A=0
11. 1**1**                                       B-A=1
12. 1**2**                                       C-A=2
13. 1**3**                                       D-A=3
14. 1**4**                                       E-A=4
15. 1**5**                                       F-A=5

Currently, due to the error by the Subtractor from 0-3 .. I cannot make a circuit to fill in 0-9 signals with proper Outputs.

Here is the link to the [file](https://temp-file.org/NrE5Wl1jVtvC7RJ/file) so you can better understand the problem yourself.. Just keep in mind that the file is uploaded temporarily and will be deleted after 1 Month.

davidhutchens, 2023-08-17:
A 4-bit add operation, when viewed in hex (meaning a number between 0 and 15 inclusive), does its operation mod 16. Thus, 9 + 9 is (18 mod 16) which is 2.

The wrap around works with subtraction also. Thus 0 - 1 is the same as 16 - 1 which is 15.

So 0-A is 16 - 10 which is 6. Note this requires a "borrow" and the borrow out (labeled b out) is active. It might be possible to use that output in your circuit.

There is no input for which the subtraction "cannot be done". It is defined for all inputs. The subtract is not a BCD component. It is a general purpose base 2 component.

The subtract circuit in your example is working as intended.

maehne, 2023-08-17:
As a complement to @davidhutchens good response, I would like to draw your attention on the `Binary to BCD` and `BCD to seven segment` components in the `BFH mega functions library`. Converting binary to BCD can be achieved with low effort, e.g., using the [Double Dabble algorithm](https://en.wikipedia.org/wiki/Double_dabble). I will close this issue as answered.

closed 2023-08-17 by maehne

Genious28, 2023-08-17:
@davidhutchens 
> A 4-bit add operation, when viewed in hex (meaning a number between 0 and 15 inclusive), does its operation mod 16. Thus, 9 + 9 is (18 mod 16) which is 2.
>
> The wrap around works with subtraction [...]

@maehne 
> As a complement to @davidhutchens good response, I would like to draw your attention on the `Binary to BCD` and `BCD to seven segment` components in the `BFH mega functions library`. Converting [...]

I see, and i tried making use of the Borrow Pin but i couldn't.. It gives me an idea, for a feature to add to the Subtractor/Adder. 
- To make a Wrap Around Pin, when 1, should Wrap. And when 0, should not Wrap.
- Also, maybe i don't understand the Borrow very well, but it would be handy if it could take Hex Values. So that you allow (or maybe force) the component to borrow more than one at the same time if you want to. A value that could be changed depending on the Input using a different Circuit.

As for Double Dabble Algorithm, thanks alot for that! i really did not understand how to use the Binary to BCD Converter without that. I'm not qualified to help you out with this but please update the Help Library with the new components. It would help out alot to understand the new Components because i was expecting to see a All the Input Binary Bit Pins as well as all of the Output BCD Binary Pins.
But hey, maybe you can add a feature to switch such components between taking/giving each individual Bits or taking/giving a Data Line, both as input and output respectively. But honestly, the Splitter can help you do that anyways so not really an extremely important thing to add.
- And a Bug i noticed, again maybe i'm wrong, but the Frequency Display on the top Left is not always Displaying the Correct TickRate/Frequency.

davidhutchens, 2023-08-18:
The adder and subtractor are modeled after very common hardware components. We are not likely to change their behavior because they map directly to standard hardware.

The frequency displayed in the upper left is reporting the actual speed of the simulation. It will typically be less than the selected clock rate because the simulation can only run as fast as your hardware can process the signals. Since we allow a large selected clock rate, the displayed rate is sometimes significantly less than (as in a small fraction of) the selected clock rate.