logisim-evolution/logisim-evolution #1834: please include a machine code inserter for of inserting risc-v machine codes

halvbanan, 2023-08-24:
i'm using rars to insert manually the machine codes through bus transactions and im getting tired of it, there is already that load function on the ram circuit so i want that on either the assembler for machine code/hex codes, or the memory simulator. i beg of you

maehne, 2023-08-24:
It is not clear to me, what enhancement you actually want. Please, explain in more detail what functionality you want to be added. Be aware that it should be sufficiently generic to work also for the other supported ISAs. You seem to be aware of the assembler window. You can also write RISC-V machine instructions into the memory be loading an ELF executable.

maehne, 2023-08-24:
Thanks for the clarification! @BFH-ktt1: Could you maybe have a look and comment?