logisim-evolution/logisim-evolution #1840: Delay component

prlaba, 2023-08-31:
One issue with L-E (and the original Logisim) is that it will not uncover possible timing issues in a circuit, since every component in the circuit has the same propagation delay.

When simulating a circuit, it would be better if the simulated components' propagation delays reflected their actual components' delays.

While it would be great if every L-E component had a _Delay_ attribute, a more reasonable approach might be to simply add a new _Delay_ component with a settable _Delay_ attribute.  The Digital logic simulator provides such a component.

The Delay component's Delay attribute assigns a number of "delay units" to the component (the default is 1). A Delay component is typically placed immediately after a component's output to reflect that component's total propagation delay (since the component itself has a delay of 1, the Delay component would add its delay to the component's delay).

This is important when using components that have long propagation delays, such as ROMs.  ROMs typically have propagation delays of 70-150 nsecs compared to 15 nsecs for a logic gate.  Having a Delay component that could add 70-150 delay units to a ROM component would allow a designer to more accurately simulate those components' delays. 

I'm not sure how this component would affect L-E's simulation engine.  Hopefully it wouldn't slow down the simulation in real time.

prlaba, 2023-08-31:
> I reckon the usefulness of such an enhancement. It would also allow illustrating some technology aspects in Logisim-evolution. Logisim's simulation engine and its shortcomings are documented to some extend in the "Value propagation" chapter of its "Guide to Being a Logisim User" (Help -> User's Guide). Are you aware of the Project -> Options... -> Simulation -> Add noise to component delays setting? Maybe, this can be a starting point for investigation what such an enhancement would entail.
> I think that this enhancement requires some investigation and willingness to dive into the guts of Logisim's simulation engine. Whether this enhancement is feasible is open. Carl Burch actually started to rewrite an educational logic simulator from scratch, because there were limitations (e.g., supporting inout ports), which he felt he was not able to overcome otherwise.
> PRs in this direction are welcome, but we currently have ressources to commit to this feature.

Yes, I’m familiar with the Add noise to component delays feature, but that doesn’t let you assign a specific delay time to a component.

The Digital simulator is not derived from Logisim, so it’s able to overcome some of Logisim’s (and L-E’s) shortcomings.  Of course, it has some of its own shortcomings.

Like many of my others, this was more of a wouldn’t-it-be-nice suggestion.  Feel free to close the issue.

Paul

maehne, 2023-09-04:
I'll keep this issue open so that this enhancement can be further discussed and people can show there interest in the feature. Besides a delay component, it might be also nice to configure the input-to-output delays for the basic logic gates – ideally in function of the number of their inputs.

prlaba, 2023-09-05:
Thanks,

Ideally, _every_ component would have a Delay attribute that would let you set the number of delay units (default 1).

I'm not sure how useful it would be to provide a Delay attribute only for the basic logic gates, since they have similar propagation delay times and are faster than all of the other components.  It's those other components, like ROM, RAM, counters, etc. that would benefit from the attribute.

maehne, 2023-09-05:
I agree such a delay attribute is useful not only on the gate level. If such a feature gets implemented, there should be some utility functions/classes, which facilitate adding these delays easily to circuit components modelled using Java code.

maspeir, 2024-08-01:
I'm dealing with this issue right now. I'm working on a simple clock circuit utilizing the built-in 74xx192. I cannot use the carry signal from the 10's counter as it is gated so that the counters clear when it reaches 6 and the carry pin only gets set with the counter goes past 9.  I tried connecting the clear signal as the carry flag, and this would work fine in real life, but the signal happens so fast that it doesn't propagate. As such, when the seconds counter goes past 59, the minutes counter does not increment. Just a tiny delay would fix this.

maehne, 2026-09-04:
@hewzhew: Your proposed delay component may be a way to test its usefulness to resolve combinatoric loop issues. IMHO, it should remain a simulation-only component and its documentation should clearly state that it is a kludge to overcome limitations of the simulation engine.

Before it gets merged into main, it should receive testing by @prlaba and it would be nice if he could share his test cases with us during the review process. Only once we gained some experience with it, we can decide whether it makes sense to officially add this component to Logisim-evolution.

hewzhew, 2026-09-05:
Thanks, that scope makes sense to me. Any prototype I prepare will remain simulation-only and will document explicitly that it is a workaround for simulation-engine limitations, not physical or circuit-wide timing modelling. Before proposing it for merge, I will ask @prlaba to share relevant test cases and test the prototype during review.

kailando, 2026-09-06:
> I reviewed the current propagation path before proposing an implementation. The simulator already has the scheduling primitive needed for a standalone component:
> 
>     * `InstanceState.setPort(..., delay)` schedules an event at the current internal simulation time plus the requested positive delay.
> 
>     * Pending changes from the same component/output are not cancelled when a later input change arrives; they remain serially ordered in the queue. Thus a simple pass-through component would currently have transport-like behavior and would preserve a pulse shorter than its configured delay.
> 
>     * These values are internal propagation steps, not calibrated nanoseconds. The existing user guide also describes built-in component delays as somewhat arbitrary.
> 
> 
> This suggests a bounded first contribution that does not require a simulation-engine rewrite:
> 
>     1. Add a standalone **Delay** component to the Wiring library with one input and one output.
> 
>     2. Persist **Data Bits** and a positive **Delay (simulation steps)** attribute, defaulting to 1.
> 
>     3. Propagate the current input value to the output using the configured delay, preserving UNKNOWN and ERROR values unchanged.
> 
>     4. Document explicitly that this is a simplified simulation aid, not physical timing analysis.
> 
>     5. Add focused tests for width/value propagation, exact step ordering, preservation of a short pulse, attribute persistence, and the default value.
> 
> 
> I would leave per-component delay attributes and a shared API for all Java components to a later change, after the standalone contract has been exercised. I would also initially treat the component as simulation-only (no claim that an FPGA/HDL flow can realize a propagation delay); if preferred, its HDL behavior could instead be discussed separately.
> 
> Would this standalone, simulation-step-based first slice be acceptable?

Why not make it act (and possibly look) like a buffer, just with a new "Delay" attribute along with "Output Value", "Label", "Label Font", "Data Bits", and "Facing"

hewzhew, 2026-09-06:
That is a sensible design. I would keep it as a distinct simulation-only Delay component, but model its appearance and attributes after Buffer: Facing, Data Bits, Output Value, Label, and Label Font, plus the configurable Delay in simulation steps. Keeping a separate component ID avoids changing the behavior of existing Buffer instances, while making the intended pass-through role familiar.

prlaba, 2026-09-06:
If it looks and acts like the existing Buffer component, would it be easier to simply add the Delay attribute to that component?  Since Delay defaults to 1 the new attribute would be backwards compatible.

Paul

maehne, 2026-09-06:
I agree that extending buffer with an additional attribute to set a delay (defaulting to 1) sounds like a sensible backwards-compatible option. One thing we might consider in addition to the delay length is whether it behaves as an inertial or transport delay (if implementable with reasonable effort).

kailando, 2026-09-06:
> I agree that extending buffer with an additional attribute to set a delay (defaulting to 1) sounds like a sensible backwards-compatible option. One thing we might consider in addition to the delay [...]

Probably transport. If inertial is wanted, something along the lines of a capacitor should be added.
Also, q: Would we measure it in clock oscs, app ticks, frames, what?

prlaba, 2026-09-06:
Yes, I think transport. Given that’s how the current Buffer component behaves, adding a Delay attribute shouldn’t change that behavior.

Paul
On Sep 6, 2026 at 6:02 PM -0400, logisim-evolution/logisim-evolution ***@***.***>, wrote:
> One thing we might consider in addition to the delay length is whether it behaves as an inertial or transport delay (if implementable with reasonable effort).

davidhutchens, 2026-09-07:
Let me give a quick overview of how the simulator works and what the delay does.

The simulator works by maintaining simulation events in a priority queue called the event queue organized by the delay. Every action has a delay whether it be generated in a gate, wire, arithmetic component, or any other component. All delays are at least 1. Different components have different delays. Many have some notion of relating to the hardware delays. Many just say 1 or a small number even if the actual hardware would take much longer than a NAND gate.

For example, a NAND gate has a delay of 1 (as does all the other basic gates). Addition has a delay of `2 * numberOfBits * PER_DELAY`. PER_DELAY is a constant used in the arithmetic components and is set to 1. So a 64 bit addition would actually have a delay of 128. The Multiplier's delay is calculated as numberOfBits * (numberOfBits + 2) * PER_DELAY. So a 64 bit multiply would have a delay of 64*66*1. But the floating point divide has the same delay as integer addition!

So lots of components have delays that are greater than 1, but many of them could be improved. For example, real hardware carry-lookahead adders do not take time proportional to the number of bits (as a ripple-carry adder would), but rather time proportional to the log of the number of bits.

Changing a given components delay is a very simple code change. Just a constant or expression. But making all the delays make relative sense would take a lot of analysis as there are so many components. And the best we could do is a rough approximation.

The simulator (actually its propagator) has a loop that processes the next group of events (whose delay is now up) in an iteration. One iteration is a simulation step. Each step generates zero or more new events with associated delays and adds them to the event queue.

So there is only one kind of delay in the simulator. There is no notion of voltage levels, capacitance, resistance, etc.

hewzhew, 2026-09-07:
Thanks, this is very helpful and corrects the original premise that all components have the same delay. I would therefore scope the prototype to a positive integer **Delay (simulation steps)** attribute on Buffer, defaulting to 1 and using the existing transport-like event-queue behavior, without introducing a separate physical timing model.

One remaining question is HDL behavior: since Buffer currently supports HDL generation, I think a non-default delay should be reported as unsupported rather than silently ignored. Would that be the preferred behavior?

kailando, 2026-09-07:
> Let me give a quick overview of how the simulator works and what the delay does.
> 
> The simulator works by maintaining simulation events in a priority queue called the event queue organized by the delay. Every action has a delay whether it be generated in a gate, wire, arithmetic component, or any other component. All delays are at least 1. Different components have different delays. Many have some notion of relating to the hardware delays. Many just say 1 or a small number even if the actual hardware would take much longer than a NAND gate.
> 
> For example, a NAND gate has a delay of 1 (as does all the other basic gates). Addition has a delay of `2 * numberOfBits * PER_DELAY`. PER_DELAY is a constant used in the arithmetic components and is set to 1. So a 64 bit addition would actually have a delay of 128. The Multiplier's delay is calculated as numberOfBits * (numberOfBits + 2) * PER_DELAY. So a 64 bit multiply would have a delay of `64*66*1`. But the floating point divide has the same delay as integer addition!
> 
> RAM, which began this issue, has a delay of 10.
> 
> So lots of components have delays that are greater than 1, but many of them could be improved. For example, real hardware carry-lookahead adders do not take time proportional to the number of bits (as a ripple-carry adder would), but rather time proportional to the log of the number of bits.
> 
> Changing a given components delay is a very simple code change. Just a constant or expression. But making all the delays make relative sense would take a lot of analysis as there are so many components. And the best we could do is a rough approximation.
> 
> The simulator (actually its propagator) has a loop that processes the next group of events (whose delay is now up) in an iteration. One iteration is a simulation step. Each step generates zero or more new events with associated delays and adds them to the event queue.
> 
> So there is only one kind of delay in the simulator. There is no notion of voltage levels, capacitance, resistance, etc.
> 
> Note that there is no clear relationship between simulation step speed, delays, simulator tick speed, and the real-time clock. The simulator ticks will not advance until the simulation event queue is emptied. So you will not see simulation errors because your logic takes too long to complete. That is beyond the capability of the simulator.

Appreciate it, @davidhutchens ! Never knew how Logisim-evolution measured time

maehne, 2026-09-07:
> Thanks, this is very helpful and corrects the original premise that all components have the same delay. I would therefore scope the prototype to a positive integer **Delay (simulation steps)** [...]

IMHO, the set output delay shouldn't have an impact on the generated HDL code. Also when VHDL/Verilog are synthesised, annotated delays are simply dropped. We might consider adding a warning, but I have my doubts about its usefulness.

Regarding my earlier comment regarding transport delay and inertial delay: VHDL supports both concepts and the difference is that in case of inertial delay (the default) glitches with a pulse width smaller than the set delay are swallowed, i.e., don't show up on the output. This is the typical behaviour exposed by gates. Transport delay simply delays all input changes leading to a phase shift of the signal.

However, after @davidhutchens comment on how the simulation engine of Logisim-evolution works and does not provide a clear link between simulation steps and simulation ticks, I am thinking that it is kind of pointless to provide this configurable delay component, as it won't really address the originally requested functionality by @prlaba. So I am tending towards closing this issue as won't fix. What do you think @davidhutchens and @BFH-ktt1?

prlaba, 2026-09-07:
I appreciate hearing the details of L-E’s internal sequencing.

Let me offer why I think a Delay component (or Delay attribute added to the Buffer component) would be useful in my use case.

My circuits need pulses with different widths. I started with the standard AND + NOT gates wired together to create a basic pulse.  But I needed a wider pulse, wide enough so the pulse remained active while other parts of the circuit(s) changed. To do that I chained together more NOT gates, sometimes as many as 81 of them, to get the desired result.  I encapsulated all these gates in a custom Pulse circuit.  In my actual circuit I use 74HCT221 Monostable Multivibrator ICs to generate these pulses).

My thought was that, with a delay component, I could simply place that component after the NOT gate of my pulse circuit and specify the desired delay, effectively doing what all those added NOT gates were doing.

I obviously have only an outsider’s view of L-E.  I’m not even sure implementing my suggested Delay component or attribute would do what I assume it would.

My project is built around digital gates and other ICs, each with their own propagation delays. While I don’t expect any digital logic simulator to fully replicate the timings of real components, I’m hoping a Delay component or attribute would help close that gap.

Paul

davidhutchens, 2026-09-07:
@prlaba: Thank you for the explanation of your use case.

I don't think many users would have a need for a delay component, but I do think it would work as well as a series of NOT gates in your case. It would also be slightly more efficient since it would skip all the intermediate simulation events. The very minor downside is adding an additional attribute to a seldom used component.

So I think we should add the delay attribute to the Buffer component, with a default of 1. I would make it an integer attribute with 1 as the lower bound and a reasonably large upper bound, perhaps 1000.

I have insufficient knowledge of the HDL side to offer an opinion. So I leave that to @maehne.

kailando, 2026-09-08:
How about adding a Pulse and Delay on buffer? First, Delay (transport) gets applied on the signal. Then, after that, Pulse (inertial) is applied.

maehne, 2026-09-28:
@BFH-ktt1: What do you think about this issue?

If we are going to implement it, I currently would opt for the least intrusive change by offering a Delay attribute to the Buffer component ideally with inertial delay semantics, but I can live with transport delay if it is simpler. As long as Logisim-evolution has not a cleaner representation of simulation time, I think it is of not much value to invest too much effort in this feature request.

I would be also fine with marking this issue as won't fix.