logisim-evolution/logisim-evolution #1841: RAM component's Output Enable pin only works with bidirectional bus

prlaba, 2023-08-31:
I have a RAM component configured with separate data input and output pins.

When configured that way, the _Output Enable_ pin has no effect; the RAM's output pins always contain the value of the currently addressed cell.

If the RAM component is configured with bidirectional data pins, the _Output Enable_ pin works as expected: when 0 the data pins are high-Z; when 1 the data pins contain the value of the currently addressed cell.

I would expect the Output Enable pin to work the same way when the RAM component is configured with separate data inputs and outputs.  But the data outputs are never in a high-Z state, even if the _Output Enable_ pin is 0.

The workaround is to add a Controlled Buffer component behind the RAM's data outputs pins and use the buffer's control pin to enable or disable the RAM's data output pins.

figurantpp, 2023-09-07:
This seems a little bit odd to me.

While using a RAM configured with separate data pins, when the output enable pin is zero, the behavior I actually get is that the RAM keeps emitting the value of the last/currently emitted output value, (so it locks in with whatever the output data pin is emitting right now) until the Output Enable pin is once again enabled (and the component is clocked, if synchronous). 

This behavior I get is what I'd expect from the component, but perhaps a 'Low Output Enable Behavior: Zero | Lock' builtin option could be useful, if not convoluted?

maehne, 2023-09-08:
@mbaillif: Could you maybe have a look on this issue, as you wrote the updated [documentation](https://www.baillifard.com/logisim/en/html/libs/mem/ram.html)?

mbaillif, 2023-09-12:
I'll look into it as soon as possible.