logisim-evolution/logisim-evolution #1850: Does it support download into ALTERA MAX10?

top58525785, 2023-09-05:
Hello
everybody
Does logsim support download into MAX10？I add FPGA model is 10M08SCM153

when i download program into fpga, It shows error as below:

Info: Command: quartus_map logisimTopLevelShell --optimize=area
Error (125095): Part name 10M08SC153 is invalid
Info (125063): set_global_assignment -name DEVICE 10M08SC153
Error (281000): Part name 10M08SC153 is illegal

So how to add a FPGA BOARD  and then can download? THanks

maehne, 2023-09-08:
Yes, MAX10 is supported. However, you need to use the [full part name](https://www.mouser.ch/datasheet/2/612/m10_overview-2401081.pdf), as when you select it in Quartus Prime, when creating your HDL project targeting the board there. There seems to be one letter and one digit missing to describe your board's FPGA fully for the mapping step.

top58525785, 2023-09-12:
Thanks for your reply
Could you help for an example for missing part
Thanks again

发自我的iPhone

maehne, 2023-11-29:
Quartus Prime offers two variants of the MAX 10 10M08SCM153 part:

* 10M08SCM153C8G
* 10M08SCM153I7G

You have to check your dev board or its schematics to know which is the one, you need to choose.

maehne, 2024-05-15:
Closing as answered due to no further activity. Feel free to reopen in case of follow up questions to this topic.

closed 2024-05-15 by maehne