logisim-evolution/logisim-evolution #1852: FPGA Board Editor Properties windows does not close with 'x'

aplq, 2023-09-08:
Using:
Logisim Evolution 3.8
Windows 11 Home build 22621

On the FPGA Board Editor when clicking on an item, such as a pin, the property menu comes up in another window. The issue is that in this window the 'X' in the top right does not do anything. Expected functionally was to close the sub window.
This seems to happen regardless of the item type. Tested with: Pin, portIO, LED Array, 7 seg, and DIP switch.

Button in question is highlighted in yellow:
[image]

maehne, 2023-09-08:
I can confirm the issue. Closing the window should be equivalent to clicking on “Cancel”.

aplq, 2023-09-11:
It is the same behavior with the "Configure FPGA" menu

aplq, 2023-09-27:
Function responsible for menus is located in src/main/java/com.cburch/logisim/fpga/gui/FpgaIoInformationSettingsDialog.java
Functions: getSimpleInformationDialog and getFpgaInformation

One solution is to set the 'x' button listener to just click the 'cancel' button using `cancelButton.doClick();`. I have a fix working that way, but I fell that there is likely a better way. The functions may need to be refactored.

maehne, 2023-09-28:
Thanks for the analysis and initial idea to fix! Feel free to propose it as a PR and then further refactorings can be discussed. You may also mark potential areas for improvements in the code using JavaDoc comments.

PR #1928 (2024-01-18): Fixed window close button on some FPGA windows
This resolves #1852
closed 2024-01-18 by BFH-ktt1