logisim-evolution/logisim-evolution #1866: Release schedule

figurantpp, 2023-09-12:
So, logisim evolution is currently at version  3.8.0, as it has been like this for a while. However, the previous major release, 3.7.0 has had two sub-releases within the same year, 3.7.1 and 3.7.2. Is logisim going to have a 3.8.1 minor release, or is the current plan to get more things done before release 3.9?

BFH-ktt1, 2023-09-14:
@figurantpp : I plan a new release in the coming days. Unfortunately I am quite busy for the moment.

cosineblast, 2024-01-12:
Are there any milestones to be met before the next release of the project? My university tends to use this project, and some of the bugfixes and features implemented in master can be handy this semester so we might end up using the nightly releases.

Anyhow, are there any milestones to be met before 3.8.1 is released?

closed 2024-08-21 by cosineblast