logisim-evolution/logisim-evolution #1867: Issue with VHDL conversion

DevGosho160, 2023-09-12:
[image]

I'm trying to convert my .circ files to .vhd files using the FPGA Commander. I am getting a fatal error where it tells me multiple sheets are named "generateMultiple". However there is only one sheet I can see in the circuits's menu to the left.

BFH-ktt1, 2023-09-14:
@DevGosho160 : could you please attach the circuit as a zip-file, in this case we can debug. Some hints that may solve the problem:

1) Save the file, restart logisim, load the file and try again, or
2) rename the sheet 'generateMultiple' and try again

Hope this helps

BFH-ktt1, 2025-06-01:
closing this issue as longtime no action.

closed 2025-06-01 by BFH-ktt1