logisim-evolution/logisim-evolution #1871: Multiple VHDL export issues with flip-flops and buses

DarkAyron, 2023-09-18:
When exporting to VHDL, all registers are uninitialized. When creating a circuit with feedback from flip-flops or using counters without connecting the load input, there's no way to get into a defined state. Some synthesizers remove these circuit part, causing the core to malfunction. Also S-R flip-flops behave different in simulation and HDL. When both inputs are 1, they stay at their value (as documented), but in HDL the turn to 0 (reset has precedence).

Also unused signals of a bus are mapped to 'OPEN' in the style of
`somebus(5) => OPEN`
this is not valid in VHDL.

maehne, 2023-09-18:
Please provide reproducers for your above issues in form of one or several (zipped) `.circ` files. Ideally, please also show relevant snippets of the wrongly generated VHDL code. It will help us to review your bugfix PR.

DarkAyron, 2023-09-18:
Here is a file, that contains .circ-files and the produced VHDL-code reproducing the bugs. When simulating the testbench (tb.vhd) for d.circ, it will always output 0 for Q, because the signal stays at undefined state. The removal of such circuits is a bit difficult to reproduce, because synthesis is a complex process. When changing line 17 of D_FLIPFLOP_behavior.vhd to
`SIGNAL s_currentState : std_logic := '0'`
it'll behave as expected.

sr.circ produces the following line in the S_R_FLIPFLOP_behavior.vhd:
`s_nextState <= (s_currentState OR s) AND  NOT (r);`
This gives the r input precedence. Correct would be:
`s_nextState <= (s_currentState AND s) OR ( NOT (r) AND s) OR (s_currentState AND  NOT (r));`

The last issue about the bus mapping can be seen in the result of shift.circ. This is the output when trying to analyze main_behavior.vhd:
```text
shift/main_behavior.vhd:90:33:error: cannot associate individually with open
                 q(0)        => OPEN,
                                ^
shift/main_behavior.vhd:91:33:error: cannot associate individually with open
                 q(1)        => OPEN,
                                ^
shift/main_behavior.vhd:92:33:error: cannot associate individually with open
                 q(2)        => OPEN,
                                ^
shift/main_behavior.vhd:93:33:error: cannot associate individually with open
                 q(3)        => OPEN,
                                ^
shift/main_behavior.vhd:94:33:error: cannot associate individually with open
                 q(4)        => OPEN,
                                ^
shift/main_behavior.vhd:95:33:error: cannot associate individually with open
                 q(5)        => OPEN,
                                ^
shift/main_behavior.vhd:96:33:error: cannot associate individually with open
                 q(6)        => OPEN,
                                ^
shift/main_behavior.vhd:97:33:error: cannot associate individually with open
                 q(7)        => OPEN,
                                ^
ghdl:error: compilation error
```
[1871.zip](https://github.com/logisim-evolution/logisim-evolution/files/12648557/1871.zip)

maehne, 2023-09-18:
Thanks for the reproducer. Which Logisim-evolution are you actually using? On which platform? Which logic synthesis tool? Have you tried with the [nightly builds](https://github.com/logisim-evolution/logisim-evolution/#nightly-builds)?

BFH-ktt1, 2023-09-18:
There are here some issues:
1) It is normal that registers are uninitialized, as "real hardware" will start-up the registers with a random value. It is only "possible" to put your system in a "defined state" by using a reset signal (being it a POR-circuit or button). It is clear that in case of an FPGA, by uploading the bit-file, registers can be initialized, but this means that in case you want to "restart" your system, you have to reload the bit-file, as otherwise the value of a given flip-flop could be '1' where the "startup value" should be '0'. This is the reason that I will not merge a PR, that puts initial values on whatever memory element (besides from ROM's), as it is bad design practice.
2) Concerning the "bit-map" to "Open", this is correct VHDL. When ghdl does not accept it, this might still be a "bug" in ghdl, and should be reported there. Also here I have problems accepting PR's.
3) Concerning the set-reset latches, there are here several problems, namely:
  a) In real hardware these latches most probably start to oscillate when both set and reset are activated, this should be avoided when possible.
  b) Latches are very poorly supported in FPGA's, meaning that it might even be that they do not function as specified in VHDL/Verilog due to the mapping on the FPGA-fabric. They should be avoided.

Hope this helps.

DarkAyron, 2023-09-19:
I tried the latest git verstion (1c7f81c3d7db713850e6ba978b1d3a0d64d5a571) of logisim on linux, the synthesis toolchain is ghdl, yosys and icebox for the ice40hx1k.

1. Concerning real hardware, as far as I noticed, when the core bitstream is loaded, all latches are initialized in reset state. If you initialize latches with '1', yosys inserts inverters, other synthesizers ignore it. Almost no FPGA supports latches with asynchronous preset or reset.
2. IEEE Std 1076-2008 section 6.5.7 says: 
> ... every scalar subelement of the explicitly declared interface object shall be associated exactly once with an actual (or subelement thereof) in the same association list, and all such associations shall appear in a contiguous sequence within that association list. Each association element that associates a slice or subelement (or slice thereof) of an interface object shall identify the formal with a locally static name.

So, partial maps to open are not correct. Vivado also behaves like ghdl and complains about partial maps to open.

3. Latches are a basic building block in FPGAs (that's why the latch and the register are in the toolbar). All the other types of flip-flops can be derived from latches by adding some gates. Logisim implements the S-R flip-flop to keep its state when both inputs are active, in contrast to the J-K flip-flop, witch toggles in that case.

DarkAyron, 2023-09-19:
I've discovered further issues with open maps. It still produces invalid VHDL code, when outputs between others left open. As a workaround placing a "Do not connect" on unused outputs creates dummy signals, making perfectly valid VHDL code. I'll create another PR only covering the S-R flip-flop issue.

maehne, 2023-09-20:
> [...]
>
>     2. IEEE Std 1076-2008 section 6.5.7 says:
>
>
> So, partial maps to open are not correct. Vivado also behaves like ghdl and complains about partial maps to open.
> [...]

@DarkAyron: Please, create a separate issue regarding partial maps to `open`. It should not be mixed with the discussion around the VHDL code generated for flip-flops and latches.

DarkAyron, 2023-09-20:
I've already created a separate issue regarding the flip-flops. I also noticed, my patch regarding the partial maps to open caused other issues resulting in non-conforming vhdl-code (removing the open maps results in a non-contiguous sequence). That's why I closed this. I'll look after this and submit a working PR when I've time.

maehne, 2023-09-20:
> I've already created a separate issue regarding the flip-flops. 

You mean issue #1874. 

> I also noticed, my patch regarding the partial maps to open caused other issues resulting in non-conforming vhdl-code (removing the open maps results in a non-contiguous sequence). That's why I closed this. I'll look after this and submit a working PR when I've time.

Then, let's reopen this issue to keep track of the partial maps to `open` issue.

DarkAyron, 2023-09-20:
As far as I've found out, this only affects the shift register. It's the only component, which exposes the elements of the output bus as individual ports. Other components, e.g. the counter only have the whole bus as a single port. For the shift register this is not possible, because it supports multi-bit, turning each stage into a bus. Maybe this could be solved by creating intermediate signals for the output of each stage.

maehne, 2024-08-16:
@BFH-ktt1: Could you have a look specifically at the shift register case?

BFH-ktt1, 2025-06-01:
@DarkAyron and @maehne : The problem with the shift register is that I wanted to create one instance that is parameterizable, but unfortianately this is not possible in VHDL due to the bit-index OPEN not supported option. I'm looking into this problem now by generating for each individual shift-register an own description, although not very efficient I see no other option to it.