logisim-evolution/logisim-evolution #1874: S-R Flip-flops behave different in simulation and HDL

DarkAyron, 2023-09-19:
Although the behaviour of S-R flip-flops is undefined when both inputs are '1', logisim keeps it's state in this case.

> S-R Flip-Flop: When the clock triggers, the value remembered by the flip-flop remains unchanged if R and S are both 0, becomes 0 if the R input (Reset) is 1, and becomes 1 if the S input (Set) is 1. The behavior in unspecified if both inputs are 1. (In Logisim, the value in the flip-flop remains unchanged.)

The resulting VHDL is
`s_nextState <= (s_currentState OR s) AND  NOT (r);`
This gives the reset input precedence. The correct line would be
`s_nextState <= (s_currentState AND s) OR ( NOT (r) AND s) OR (s_currentState AND  NOT (r));`

BFH-ktt1, 2023-11-03:
@DarkAyron : Can this PR be closed due to the merged PR?

maehne, 2023-11-29:
Closing this issue as resolved, as PR #1875 got merged and @DarkAyron did not react.

closed 2023-11-29 by maehne