logisim-evolution/logisim-evolution #1876: Add the ability to load multiple memories from files on the command line

cdubach, 2023-09-26:
Currently, it is only possible to load a single RAM in a circuit in the command line. Ideally, we should not limit the loading to a single RAM, and allow the loading of one or more memories (either a RAM or a ROM).

I suggest that we add the ability to specify on the command line the "label" of the memory we wish to load and allow the load argument to appear multiple times if more than one memories need to be loaded.

I have implemented this feature on my fork and I am currently testing it. Will open a pull request soon.

cdubach, 2023-09-27:
Just created a pull request (#1877) with my changes to the code to implement this.

TwinT, 2024-11-04:
This is a very useful feature for scripting. Please consider adding it on a next release.