logisim-evolution/logisim-evolution #1888: Add TTL circuits 7476 and 7493

TAHRI-Khalid, 2023-10-11:
Hello 
I'm a teacher and logisim-evolution helps me a lot in my work , I wish if you can add these circuits to TTL

- 7476 a dual JK Flip-Flops With Preset And Clear
- 7493 a 4-Bit decade and binary Counter

Thank you so much for considering my request

BFH-ktt1, 2023-10-11:
Please do not hesitate to publish a PR's in this direction, They are more then welcome! As our resources are very limited, as we are also teachers, and do this in our free time, we appreciate any addition that you can make. For the moment, unfortunately, we do not have the bandwidth. I'll keep it open as is. Please do not hesitate to help us in our effort to make logisim-evolution even better.

maehne, 2023-10-16:
@BunchOfBits: Could you maybe help with @TAHRI-Khalid request?

BunchOfBits, 2023-10-18:
I will see what I can do. I am in the middle of preparing training material though for software engineers who want to learn about cpu design, which takes up a lot of my time.
I am a bit surprised about the ttl chips though: apart from the set/reset polarity, the 7476 is identical to the built-in JK-flipflop in the Memory library. Something similar can be said about the 7493 and the Counter in the Memory library: it doesn't have the separate clock inputs for the least significant bit and the upper three bits, but I have never seen them used separately, only tied together.
So, is there any particular reason for adding them to the TTL library, other then their appearance ?

maehne, 2023-10-19:
Thanks for your feedback @BunchOfBits! We are neutral regarding, which parts should be in the TTL library, as we aren't using them at all in our teaching.