logisim-evolution/logisim-evolution #1889: Combinational Analysis Not Fully Minimizing

UsrRed7, 2023-10-12:
Working with a 4 bit input, some desired outputs aren't fully minimized.

### Some examples:
1) Data: [1, 0, 1, 1, 0, 1, 1, 0 , 1, 1, 0, 1, 1, 1, 1, 0]
Formula: bc'd+a'b'd'+a'b'c+a'cd'+b'c'd'+ab'd+ac'+bcd'
The K-map generated:
[image]

2) Data: [0011111011110111]
Formula: b'c+ab'+a'bc'+a'bd'+bc'd+cd'+ad
The K-map generated:
[image]

---
I haven't looked at the code at all, but if pointed in the right direction I may be able to help track down the reason for this issue.

maehne, 2023-10-16:
Thanks for reporting the issue. Indeed, the form is not minimal, but at least correct. We highly appreciate any help to improve the optimization algorithm. The relevant code lives in `src/main/java/com/cburch/logisim/analyze/`. PRs are always welcome!

superenginegit, 2023-11-28:
If no one is working on it, I would like to work on it.

maehne, 2023-11-28:
@superenginegit: Thanks for volunteering! No one is currently working on this issue and the related issue #1914 (which I am at the moment not able to reproduce). Feel free to open a PR as soon as you would like feedback from us. You can mark it as draft als long it is not ready for merging. For the moment, I will assign you the issue.

davidhutchens, 2024-05-22:
@UsrRed7: Could you try the latest [Nightly build](https://github.com/logisim-evolution/logisim-evolution/actions/runs/9104282361) to see if this issue was fixed?

UsrRed7, 2024-05-30:
Sorry for the delay, but I tried it out for a bit and I also wasn't able to find any more logical issues. Thanks @superenginegit for the fix!

closed 2024-05-30 by UsrRed7