logisim-evolution/logisim-evolution #1890: PORTI/O number of pin CANNOT be 1

gauchl, 2023-10-12:
Hello,

I need to have one pin inout using PORTI/O but I receive an error message instead.
logisim-evolution v3.8.0
[image]

Regards,
Laurent Gauch
EMF-Switzerland

primechaser, 2023-10-13:
This has been fixed after 3.8.0 was launched. Did you install the latest nightly build?

gauchl, 2023-10-13:
Great if this was already fixed. I will check/control with the latest nightly build. THANKS`!

Note: I am searching a solution for inout port solution able to do the job in simulation and in VHDL generation ... actually the PORTI/O is OK for VHDL generation, but KO for simulation. In an other hand, the logisim-evolution simulator is very good with tristate gate, using correctly '1' '0' 'U' 'E', with or without pullup. My idea could to have an inout port viewed by the simulator as an 'wire' (not an input port, not an output port, but a **inout wire**)

[image]

Regards,
Laurent Gauch
EMF-Switzerland

maehne, 2023-10-13:
For your convenience, you can find information on using the nightly builds [here](https://github.com/logisim-evolution/logisim-evolution#nightly-builds). We are well aware of the limitations of Logisim-evolution when it comes to tristate signals and lack of inout ports. Unfortunately, the root cause in Logisim's simulation engine is very hard to fix, as it would more or less require a rewrite. To work around the issues, it is best to split up inout signals into three dedicated signals for input, output, and output enable, as you've done in your screenshot. Use these signals in your system's interface. That Logisim circuit can be then instantiated into one "top" circuit, which adds the portio components realizing the tristate buffers for the targeted FPGA board. For simulation, create a separate circuit, which instantiates your system circuit and then wires it up to the components representing the environment/communication partners.

I hope these hints help! Please, provide feedback whether your issue is fixed in the nightly build!

gauchl, 2023-10-14:
Thank you for your comments. 
I am well aware that it is very difficult to modify the simulator if the 'Z' tristate state has not been taken into account from the start of logisim dev. This is how.

I downloaded and installed on a Win machine the logisim-evolution-nightly-907 -> I can confirm that the PORTI/O with number of pins 1 is as I wished. Great Job.

closed 2023-10-14 by gauchl

maehne, 2023-10-15:
Thanks for confirming that your problem is resolved in the nightly build!