logisim-evolution/logisim-evolution #1902: Feature Request - TinyFPGA BX synthesis

hawaclawek, 2023-10-31:
At the moment, it seems Logisim-evolution is only supporting target FPGA synthesis tools from Xilinx and Intel (Xilinx ISE, Xilinx Vivado, Intel Quartus Prime). The low-cost, open-source TinyFPGA BX boards ( https://tinyfpga.com/ ), however, feature a built-in USB bootloader that allows to load a Verilog program onto the device simply via Python script. After preparing a Python venv (See description below) all that one essentially needs to do in the target directory is to:
- Activate Python venv
- Run: apio init --board TinyFPGA-BX
- copy content of https://github.com/tinyfpga/TinyFPGA-BX/blob/master/apio_template/pins.pcf as PROJECT_NAME.pcf to directory
- Create PROJECT_NAME.v containing Verilog code (see example https://github.com/tinyfpga/TinyFPGA-BX/blob/master/apio_template/top.v )
- Run: apio build
- Run: apio upload

Our university would like to utilize the power of Logisim-evolution with regards to designing digital circuits in educational settings and combine it with the easy to use and open source TinyFPGA BX boards. However, at the moment, in the FPGA board editor window, after selecting an image of the board (e.g. the one attached that comes shipped with the Tiny FPGA BX board, see https://github.com/logisim-evolution/logisim-evolution/assets/81801454 ), one cannot select another FPGA manufacturer than Altera, Xilinx or Vivado (Tiny FPGA BX runs a Lattice ICE40LP8K FPGA).

It would be sufficient for us if Logisim-evolution would let us export a Verilog file with the pinning of mentioned .pcf file, that we can load onto the board via mentioned Python script ourselves. Is there any chance that the usage of such a "custom / open"  synthesis tool pipeline is implemented into Logisim-evolution? Other boards with similar bootloader approaches could benefit from this as well.

Ad Python venv:
- Install Python 3.6.5 and add to Path variable
- Create Python 3.6.5 venv
- Activate venv
- pip install apio==0.4.0b5 tinyprog
- apio install system scons icestorm iverilog
- pip install --upgrade --no-cache-dir "tinyprog>=1.0.23"

Ad Sources:
https://tinyfpga.com/bx/guide.html
https://github.com/FPGAwars/apio
https://apiodoc.readthedocs.io/en/stable/source/quick_start.html

BFH-ktt1, 2023-11-03:
@hawaclawek : I like this approach a lot. For your information: The nightly build of logisim already contains a route that supports the open-source toolchain (GHDL)-Yosys-nextnpr-ecp5 for the latice ECP-5 family of FPGA's (with as an example the GECKO5Education (https://github.com/logisim-evolution/GECKO5Education)). I will look into the details of the tinyfpga board. However, my time is quite limited.

hawaclawek, 2023-11-03:
Thanks for the quick reply and the good news! Looking forward to any progress on this then.

cslammy, 2024-02-17:
Seconding this request....TinyBX support would be very helpful for teaching logic and FPGA basics--TinyBX is arguably the "Arduino of FPGA"; Its affordability and simplicity makes it a good choice for students and newbies.  Thanks.