logisim-evolution/logisim-evolution #1913: Clock won't pulse by itself

AbiPrescott, 2023-11-24:
Hi. I'm having an issue: clock won't pulse by itself. I have to poke it to make it high or low. 

I am on Ubuntu Server 22.04

I am on Logisim Evolution v3.8.0.

It was installed from snap store using command: sudo snap install logisim-evolution-snapcraft

I checked my java version ($java --version), and this is what I get (which meets the requirement of having java 16 or higher):

java 21.0.1 2023-10-17 LTS
Java(TM) SE Runtime Environment (build 21.0.1+12-LTS-29)
Java HotSpot(TM) 64-Bit Server VM (build 21.0.1+12-LTS-29, mixed mode, sharing)

I'm guessing it is .jar? I found a logisim-evolution.jar in '/snap/logisim-evolution-snapcraft/46/logisim/logisim-evolution.jar'.

Thank you :)

closed 2023-11-24 by AbiPrescott

maehne, 2023-11-27:
@AbiPrescott: Are you sure that you selected `Simulate -> Auto-Tick Enabled` from the menu to activate the clock generation with the selected `Auto-Tick Frequency`?

maehne, 2023-12-04:
OK, seeing "[SOLVED: the 'enable clock pulse' button was disabled by default. I had to click it to enable. The 'enable clock pulse button' is near the play/stop button in the simulation tab]", I will mark this as invalid, as it is not a bug, but expected behaviour that auto-ticking of the clock is disabled by default, as it causes high CPU load. Auto-ticking should only be enabled when you are actively interacting / monitoring the simulation.

AbiPrescott, 2023-12-04:
@maehne Yeah, thank you, and sorry about that. I just didn't look hard enough before I opened the issue.