logisim-evolution/logisim-evolution #1914: Minimizing Expression Incorrectly

OliviaY1, 2023-11-25:
When I typed the following expression into logisim 
[image]
Logisim minimized it into the following expression:
[image]
The minimized expression is wrong b/c based on logisim's version, ALUControl1 is not 1 when ALUOp1:0 is 00.

maehne, 2023-11-28:
I cannot reproduce your issue: When I enter the expression `ALUControl[1]` from the first screenshot, I obtain first the unmodified expression in mathematical (Boolean algebra) notation. When I switch to the Minimized tab and back to the Expression tab, I get the correctly simplified form `~ALUControl[1] + ~ALUControl[0] * Funct[1] + ~ALUControl[0] * ~Funct[0]` and not your expression with the additional parentheses in the second screenshot. I am using the current development head of Logisim-evolution. Please check with the last [nightly build](https://github.com/logisim-evolution/logisim-evolution#nightly-builds). If you can still reproduce the issue, please provide the problematic expressions in text form and not as screenshot so that we can be sure that we enter your expressions identically.

matheusbdq, 2023-12-04:
i am having the same problem too.

This is the truth-table that i was trying to simplify.
[image]
and the software gives this simplification
[image]
yet when using an old version of logisim it gives the correct simplification
[image]

the newer version is 3.8.0 and the older version that works is 2.7.1

Here is the truth-table file if anyone wish to test it
[SAIDA CONTADOR.txt](https://github.com/logisim-evolution/logisim-evolution/files/13551852/SAIDA.CONTADOR.txt)

davidhutchens, 2023-12-04:
@matheusbdq: In the second (2.7.1) part, you named the variables A, B, C, X with output KC. It gave you the answer X. In the first (3.8.0) part, you named the input variable ABCX[3..0], which is a single 4-bit variable, and the output KD. It gave you the answer ABCX[0] which corresponds to X in the second part. So both programs give you the same answer, just in different form because you stated the inputs differently (one case is 4 variables, the other a single 4-bit variable).

matheusbdq, 2023-12-04:
> @matheusbdq: In the second (2.7.1) part, you named the variables A, B, C, X with output KC. It gave you the answer X. In the first (3.8.0) part, you named the input variable ABCX[3..0], which is a [...]

I just tried what you told me and it works! Thank you friend, i have at most 2 hours of experience in using logisim so i am quite embarrased.

maehne, 2023-12-05:
Closing this issue as resolved since @matheusbdq example is correctly treated, I cannot reproduce the issue reported by the OP @OliviaY1, and didn't receive any feedback. @OliviaY1 feel free to reopen if the issue persists on your side and you can provide more detailed instructions allowing us to reproduce the problem on our side.