logisim-evolution/logisim-evolution #1916: The equivalent circuit does not work.

RedErr404, 2023-11-28:


davidhutchens, 2023-11-28:
I may be missing something, but it looks to me that you have a race condition with which changes arrive first. In going from 3 to 4, if the first two bits drop before the third bit rises, that may cause the output from the AND to rise giving the OR a short pulse of a 1 causing its output to rise which will trigger a reset. That reset is immediate.

This is why it is recommended that you always clock the memories to ensure all inputs have settled before storing a value. A reset should not be used for logic.

maehne, 2023-11-28:
I agree with @davidhutchens: Your design doesn't follow recommended design approaches for sequential circuits, as you misuse the asynchronous set/reset inputs of the D-flip-flops for other purposes than the power on reset. Flip-flops should be always clocked by the same clock signal and their next state should be determined via the D input to guarantee synchronous behaviour. I will close this issue as invalid. Feel free to reopen if you disagree.

closed 2023-11-28 by maehne

RedErr404, 2023-11-29:
What is the fundamental difference between these schemes?
[image]
[image]
This is the equivalent circuit. Or does this take into account delays between blocks?
[image]

maehne, 2023-11-29:
Please check the "Value propagation" chapter in the "Guide to Being a Logisim User" (reachable via Help -> User's Guide) for some explanations of Logisim's simulation algorithm. Indeed, gate delays are considered to some extend - though Logisim's concept of simulation time is limited. Therefore, hazards may occur due to the propagation of signal changes through the different paths from inputs to outputs traversing the different layers of gates.