logisim-evolution/logisim-evolution #1925: Program hangs when trying to optimize the circuit in the attached file

gh228df, 2023-12-12:
After pressing "optimize maxterms" programs does some output for a minute and then hangs.
here is the circuit file:
[cpufpu4x16.zip](https://github.com/logisim-evolution/logisim-evolution/files/13653940/cpufpu4x16.zip)

maehne, 2023-12-12:
Well, I am not surprised, as your 28 MiB truth table has 17 input variables and many output variables. The logic optimization is tuned for considerable smaller educational examples. @BFH-ktt1 can comment more, as he worked on the implementation of this feature. Improvements in this area are welcome in form of PRs, as long as the educational character of the tool is respected. For us, this enhancement request is of low priority.

BFH-ktt1, 2023-12-19:
I have done tests with 10-input functions. It takes in this case logisim 2 hours to determine the smallest representation. Hence with 17 inputs, be patient, I guess it might take more than 8 hours. This is not a bug, just the algorithm that exponential explodes on the number of inputs and outputs used. This cannot be fixed, for these kind of logic functions I would suggest to use "real synthesis" tools. 
Closing the issue as this cannot be fixed.

closed 2023-12-19 by BFH-ktt1