logisim-evolution/logisim-evolution #1926: Strongly holped to have a feature of Add autosaved.

dukuzu, 2023-12-15:
Without autosaving, changes made are always lost when the computer is shut down.

maehne, 2023-12-17:
Yes, this would be a useful enhancement. However, auto-saving should not happen to the opened circuit file, but rather a separate file, which gets automatically loaded by Logisim-evolution when it gets restarted after a crash. PRs in this direction are welcome!

superenginegit, 2024-01-18:
I would be interested in working on this. 

But I see a few possible ways of doing things for this.

For the autosave file. I think it is possible to either have a file with the same name as the file that is being edited in the same location, just with an appended .autosave or similar. this could then be loaded, when the same file is edited next time. Or there could be an autosave file in a central location, that would be checked for on every startup. I would lean towards the first option here, as I think it would be more practical and allow for multiple instances of the program to edit different files without interfering.

Also I see multiple ways of performing an autosave. It could either be done after every single editing action, or every x seconds (possibly a setting to choose a time interval). In my opinion the second option here is better, as after every single action could lead to performance issues, especially on larger files.

If anyone is working on this or has any opinions on this, please let me know.

davidhutchens, 2024-02-27:
I am closing this issue as completed with #1965.

closed 2024-02-27 by davidhutchens