logisim-evolution/logisim-evolution #1932: i can't import 8-bit rgb values into the ROM properly

Trixil, 2023-12-20:
i previously played an anime opening on the rgb video (https://www.youtube.com/watch?v=ZE2rrnOzTJI) using imported 24-bit RGB values. issue is, it plays super slowly and with inconsistent framerate. a 1:30 anime op took ~35 minutes to display, even at max frequency. i'd like to try using 8-bit RGB values.

when i import 8-bit rgb values into the ROM, it fails to interpret them as 8 bits and separates them into 2-bit values and associates a memory address with each of them.
[image]
thoughts?

davidhutchens, 2023-12-20:
I do not see any 2-bit or 8-bit values in your file snippet. Each grouping consists of 5 hex digits (4 bits each, for a total of 20 bits). When reading this file, LE is reading 2 hex digits (4+4=8 bits) for each 8-bit value. It is ignoring white space giving you the result that you state.

You need to provide a file where each 8-bit value is written as 2 hex digits.

You didn't ask, but looking at what you are doing, I doubt that you will notice any speedup going from 24-bit values to 8-bit values. The simulation engine is processing all the bits in parallel as is.

Trixil, 2023-12-20:
> I do not see any 2-bit or 8-bit values in your file snippet. Each grouping consists of 5 hex digits (4 bits each, for a total of 20 bits). When reading this file, LE is reading 2 hex digits (4+4=8 [...]

thanks for the reply. but in parallel? this is an rgb video component and i have to feed in each pixel color and location in one at a time. isn't this serial? i'd like to use the led matrix since that is in parallel, but it lacks color.

davidhutchens, 2023-12-21:
I am using the current dev build, so perhaps we don't match. I could go back and try your version if you tell me what it is.

My version loads all 24 bits of color and both location values on a single rise of the clock. That is what I meant by "in parallel". I did not mean that the whole screen could be loaded in parallel, just one pixel. I don't think that switching to 8-bit color would speed up the simulation much since each pixel would still be loaded in a clock tick.

I was also noting that our simulation manipulates line values (up to 64 bits) together, so each bit is not manipulated individually. Thus 24-bit values flow through the simulation at the same speed an 8-bit value would.

If you wish to try our dev build, you can find it in our [nightly builds](https://github.com/logisim-evolution/logisim-evolution/actions/workflows/nightly.yml).

davidhutchens, 2024-01-15:
Since there has not been a follow-up, I am closing this issue as answered. If there are remaining issues, please reopen this issue and explain.

closed 2024-01-15 by davidhutchens