logisim-evolution/logisim-evolution #1934: Font size mismatch in the menu and entities

gtxzsxxk, 2024-01-01:
Hello, long time no see, when playing with the `latest nightly build` I find this issue: the font size is very weird.

This is the version built in 2023-08-02

[image]

This is what looks like now:

[image]

The one with the problem was built on 2023-12-20T00:06:44+0000. I downloaded it here: https://github.com/logisim-evolution/logisim-evolution/actions/runs/7269030544,

Do you have the same problem? If this is really a problem, I mean, not only happens on me, I'd love to find out and help solve this.

davidhutchens, 2024-01-15:
I do not see this problem. Is it possible that you have changed a system font size? I'm not sure, without digging through the code, how we select the font size.

gtxzsxxk, 2024-01-15:
> I do not see this problem. Is it possible that you have changed a system font size?

I ran the two versions on the same windows 11 environment at the same time. I haven't change the system font size. So I tend to believe the problem should be attributed to the commits after 2023-08-02.

> I'm not sure, without digging through the code, how we select the font size.

OK. I would have a try.

gtxzsxxk, 2024-01-16:
This font-size problem only happened on me when I install the logisim-evolution through the windows installer provided by the latest nightly build. If I exec `java -jar logisim-evolution-3.9.0dev-all.jar`, which is downloaded from the nightly build as well, the abnormal problem goes away.

maehne, 2024-05-15:
@gtxzsxxk: Could you please try with the [latest nightly builds](https://github.com/logisim-evolution/logisim-evolution/actions/runs/8993940035) compiled with JDK 21?

maehne, 2024-05-15:
This may be a duplicate of issue #1452.

gtxzsxxk, 2024-05-16:
@maehne Hi, I tested the latest nighly build and it looks pretty good. This problem might have been fixed.

[image]

maehne, 2024-05-16:
@gtxzsxxk: Thanks for testing! So this issue seems to be resolved. I am closing this issue. Feel free to reopen in case of a regression (or check whether issue #1452 is a better fit to continue the discussion).

closed 2024-05-16 by maehne