logisim-evolution/logisim-evolution #1939: Random coordinate initialization errors

tomoish, 2024-01-08:
### Describe the bug
An error occurs with the initial values of coordinates upon startup. 
The coordinates where the error occurs are random at startup. For example, as shown in Figure 1, the error occurs at coordinates (250, 230).

[image] 
Fig 1. Error occurs at (250, 230).

[image]
Fig 2. The initial values are errors for many coordinates ((180, 140), (180, 200), (240, 160), (240, 220)).

### Reproduction Steps
The coordinates where the initial values cause an error are random, but they generally occur within the range of (200, 200) to (300, 300). 
The issue is more easily identified when using components with a large number of pins, such as TTL.

I have confirmed that errors occur at coordinates (180, 140), (180, 200), (210, 180), (240, 160), (240, 220), (250, 230), (270, 230), (280, 230) and (290, 260).

### logisim-evolution version
logisim-evolution-3.8.0-all.jar

### Java version
java 16.0.1 2021-04-20
Java(TM) SE Runtime Environment (build 16.0.1+9-24)
Java HotSpot(TM) 64-Bit Server VM (build 16.0.1+9-24, mixed mode, sharing)

davidhutchens, 2024-01-15:
I am unable to reproduce this problem. I have tried it with the 3.8.0-all.jar file and with our current head of main. I tried it with Java 16 and Java 21. I used the TTL and moved it all over the area, specifically checking the 210,130 location shown above. The only errors I see are the outputs from the NAND gates in the TTL (which are correct as no inputs are provided). I do not see the errors on the inputs as shown in your picture. I also tried the clock as shown above and did not see any problem.

Could you be more specific about what you mean by "at startup"?
 - Are you saving the circuit and loading it? If so, could you give us the circuit by zipping it and dropping into a message here?
 - Are you launching the jar and then dropping the component onto that space? Do you move it around to find more errors? This is what I tried.
 - Do the errors persist if you reset the simulation?

tomoish, 2024-01-18:
@davidhutchens 

The coordinates at which the error appears will change each time I launch using the command below, so we must look for it.
`java -jar logisim-evolution-3.8.0-all.jar`

As shown in the video below (0:26, 0:36-), an error occurs at some coordinates after launching `logisim-evolution-3.8.0-all.jar`.

Therefore, the answers to your questions are below.
* No, I didn't save and load a circuit.
* Yes, I dropped a component onto the space, and I moved it around. If I look closely at the video, the error also occured at 0:26.
* Yes, they do. Resetting the simulation has nothing to do with this error

https://github.com/logisim-evolution/logisim-evolution/assets/103555868/77ca8a55-502b-4fb8-9724-d17bcf2dd45a

davidhutchens, 2024-01-19:
@tomoish: Thank you for the answers. After more experimenting, I can now conclude that this is not random. I can reproduce the behavior.

Open a new document. Place a 7400 TTL on the canvas. Drag a wire up 2 spaces from the left top red colored output. Drag the bottom of the wire up 1 space. This results in an unconnected wire pointing to the location of the red output, so we can find it. Now select the TTL and rotate it south with down arrow key. Now drag the TTL so the upper left input is located at the space pointed to by the wire. It will show as red.

The workaround is to reset the simulation.

The problem seems to be in the disconnecting of the rotated item from the old location. I'm not very familiar with that part of the code. Perhaps @BFH-ktt1 has thoughts.

I will change the name of this issue to reflect the cause.

davidhutchens, 2024-01-21:
I have tried more experiments. The error values remain when rotating a component. But they do not remain when dragging or deleting a component. So those two actions must do something to update the value at the coordinate. If so, we should be able to fix this by applying the same logic on the rotate action.

davidhutchens, 2025-03-23:
PR #2170 fixes this problem.

closed 2025-03-26 by davidhutchens