logisim-evolution/logisim-evolution #1952: Circuit show unexpected error as a value

LaXis315, 2024-01-23:
Hi, I was trying to use this programme to study some logic circuits and while testing the multi-value input feature I encountered this problem:
when you have both bits congruent then nothing happens, otherwise an error is sent. There doesn't seem to be any overlapping of wires in the circuit, so I can't figure out what could be causing the error

[image]

davidhutchens, 2024-01-23:
Did you try resetting the circuit? Does it have the problem if saved and loaded? Could you try it in the latest version of LE, 3.8.0?

To say much more, I think we need to examine the circuit.

If the problem remains in a saved and loaded .circ file, could you zip the file and drag it into a comment here?

LaXis315, 2024-01-23:
I will carry out the required tests, but first I have to replicate the circuit as the created save file was empty. In fact, I wanted to point out this problem of the saves not seeming to work (I had made 2 for safety's sake and both were empty; there were also empty .bak files)

LaXis315, 2024-01-23:
ok, after remaking the entire circuit the error is gone. Also the savefile now work fine

davidhutchens, 2024-01-23:
I will close this issue as completed.

closed 2024-01-23 by davidhutchens