logisim-evolution/logisim-evolution #1954: SonarQube is not running properly

davidhutchens, 2024-01-26:
SonarQube is causing our build script to die on merge to main. Specifically, it is giving an error that it can't find a server at http://localhost:9000. According to the documentation at [https://docs.sonarsource.com/sonarqube/latest/devops-platform-integration/github-integration/](https://docs.sonarsource.com/sonarqube/latest/devops-platform-integration/github-integration/), we should be giving it the URL of the SonarQube Server in SONAR_HOST_URL and we should have a secret of that name with the URL as its value.

I am providing PR #1953 that has our build script check for that secret and pass it to SonarQube. The PR will silence SonarQube until a SONAR_HOST_URL secret is provided.

According to the above documentation, we should have 1) a SonarQube Server, and 2) a Github App to receive the info from the server. We might well have those things from when this was working. But I can't see them since I am not an owner of the project. If we don't have them, we will need to create them following the instructions in the above documentation.

@MarcinOrlowski might be able to tell us where they are and what URL to use if @BFH-ktt1 doesn't see them. Any other insight @MarcinOrlowski has about getting this work would be much appreciated.

maehne, 2025-06-01:
@MarcinOrlowski: Could you help us to resolve this issue?

maehne, 2026-09-07:
Thanks for your analysis @hewzhew!

@zdimension: Could you please comment on this issue, as you initially added this feature (PR #1458)? Does it work when having a SonarQube server running locally? Could you support us fixing this issue?

I am not sure whether SonarQube still offers free use for this service for open-source projects, hosting such a server on our own is additional effort. The `SONAR_TOKEN` for this projects seems to be empty at the moment. I see value in static code analysis to support the PR reviews, but we lack time to implement it ourselves unless someone is willing to support us in this matter.

zdimension, 2026-09-07:
SonarQube does still have the FOSS plan, I'll fix that this afternoon.

maehne, 2026-09-07:
Thanks @zdimension for looking into this issue!

maehne, 2026-09-28:
@zdimension: After your PR #2967 to address this issue, SonarQube messages were generated only for a few PRs. So, it seems that something is still not correctly set up. Could you please have a look?

zdimension, 2026-09-29:
Indeed, something seems wrong. I'm looking into it.

*edit:* yes; aftre 2967 I disabled Sonar's automatic analysis to instead use the CI one, but it seems the CI run doesn't work for PR coming from forks since it doesn't give them the SONAR_TOKEN value. I'm looking for ways to make that work.

zdimension, 2026-09-30:
I think our best option is to add a manual workflow that is only launchable by maintainers after having reviewed that the PR isn't adding a secret stealer in the Gradle build code. So, like, when a PR is made, regular CI is ran, and one of us can review it and after it has been verified that there is nothing fishy in the changes, can trigger the Sonar run for the PR code. What do you think?  @maehne

maehne, 2026-09-30:
Thanks @zdimension for looking into this issue. I am fine with the proposed solution to have a manual flow, which can be triggered from time to time on our repository to identify open issues. Could it be an option to have both, i.e., the manual flow to analyse in detail our code base and the automatic analysis to review the PRs without the risk of executing untrusted Gradle code?

zdimension, 2026-09-30:
It would be the optimal solution but I don't think it's possible. There's only a global setting for Automatic Analysis in Sonar, and they say that it shouldn't be used at the same time as CI analysis. We could get rid of CI analysis and only use the automatic one but then we'd just get less precise analysis all the time. I'll have a look at adding a manual workflow.