logisim-evolution/logisim-evolution #1956: UI issues and crashes on window moving

marwinhormiz, 2024-01-29:
Hi.

I recently installed logisim evolution and quickly got issues with the UI where the UI will start to mess up completely and things start to disappear. The other issue is where the program instantly crashes whenever I try to move the window. It first started with the latest version 3.8.0, then I downgraded to 3.7.2 to check if it had anything to do with the latest release, and I was still getting the same issues. I ran the program with a debugger aswell to trigger the crash whenever I just move the windows handle and to provide a backtrace for the whole callstack.

Images:

UI:
[image]
[image]

Window move crash:
[image]
[image]

Graphics acceleration in experimental is set to "Use Defaults", but have been tried with Direct 3D aswell, still same issues.

Specs:
OS: Windows 11 Pro (10.0.22631)
CPU: Ryzen 5 5600x
Graphics card: 4060 Ti
RAM: 32GB

maehne, 2024-01-29:
Your JDK 8 is too old! Logisim-evolution 3.8 requires at least JDK 16. You did not indicate which pre-compiled package you used or wherever you built Logisim-evolution yourself. I would recommend you to use the provided OS-specific native installers/packages. They include already a matching Java runtime environment. You can also use one of the recent nightly builds, which fix quite some issues of version 3.8. Please report back whether this fixes your issues.

marwinhormiz, 2024-01-29:
Oh I'm sorry, maybe I should have been more specific. I have JDK 21 according to my installed programs, but the runtime environment is 8u401, which is the latest I can find in java's official website. I don't really use Java required applications that much so hopefully I'm not confusing you with this information. I installed x86.msi and also the .jar file. When I run the executable, logisim-evolution.exe I get these issues, but if I run the .jar file immediately then everything works fine. 

[image]

marwinhormiz, 2024-01-29:
Never mind the above, I just realized I don't need the Java 8 so I uninstalled it completely. But I still have the same issues only if I run the executable but not with the jar file.

closed 2024-02-01 by marwinhormiz