logisim-evolution/logisim-evolution #1960: Do/Un-Do/Re-Do not positioning properly

miczac, 2024-02-05:
In v3.8 a Do/Do/Undo/Undo/Re-Do/Re-Do sequence the last Re-Do messes up the circuit.
I can provide a screen recording and an example file. (.zip)
[move_ctrl_z_bug.zip](https://github.com/logisim-evolution/logisim-evolution/files/14168453/move_ctrl_z_bug.zip)
This is on Mac OS 10.14.6.
Thanks a lot, Michael.

miczac, 2024-02-05:
Just checked on an up-to-date Arch Linux w/ Cinnamon: behaves exactly the same.  
HTH, Michael.

davidhutchens, 2024-02-05:
Thank you for the report. I can reproduce the problem. I ran into a similar issue a week or so ago and I was trying to figure out what is happening.

I suspect it has to do with the fact that selection / deselection is not actually recorded in the undo stack, but some actions that are recorded apply to the selected objects. This will require more research and careful thinking.

miczac, 2024-02-06:
Thanks for the quick response!
It appears the issue with what's recorded to the stack is more complex.
In this example, I just stumbled upon, _moving_ a freshly pasted object gets skipped in the undo queue.
(if I put this correctly)
You can see, _move_ doesn't appear in the undo-menu.
A second _move_ got recorded.

https://github.com/logisim-evolution/logisim-evolution/assets/1922103/5d5bc8ef-34c2-4e86-a147-803129d936d5

davidhutchens, 2024-02-07:
@miczac: I don't think that the paste drag behavior is a problem. The paste puts a selected object on the screen, but doesn't actually attach it. That way, you can drag it to where you want it without it grabbing a wire by accident and causing it to deform. It isn't until you drop it that the "paste" is actually attached to the diagram. Of course, if you do something to deselect it where it pasted, it would be attached at that point. Given that behavior, it would actually be a problem to undo the move without undoing the paste since that would attach it where it was never attached before.

maehne, 2024-08-15:
I agree with @davidhutchens analysis. Based on the screen recording, the behaviour seems to be correct and intentional. @BFH-ktt1: Do you agree?

If yes, I think that this issue can be marked as invalid and closed as won't fix.

davidhutchens, 2024-08-16:
The second issue is intended behavior. But the original issue is a bug in the interaction of selection and undo. I haven't yet come up with a clean way to fix it however.

maehne, 2024-08-16:
Thanks for the clarification @davidhutchens! Then, let‘s leave this issue open.

davidhutchens, 2024-08-22:
I have looked at this in more detail. The problem can be shown by: Create a new project. Add two gates to canvas. Select one and move it. Select the other and move it. Undo. Undo. They are now where originally placed. Redo. That works. Redo. That will move the wrong one!

The problem is that Redo is using the current selection for the move rather than the selection that existed when the action was first taken.

I have studied the code and mostly understand the issue. It will take more study to figure out the best way to fix this issue without breaking something else. I am assigning this issue to me.

maehne, 2024-08-22:
Thanks for your detailed analysis and looking further into this @davidhutchens!

davidhutchens, 2024-09-01:
I looked at all the subclasses of `Action` in `SelectionActions.java`. They all have the same problem of using the current selection when doing a `redo` (which is done by calling `doIt`, the same method that does the original operation). They are: Anchor, Copy, Cut, Delete, Drop, Duplicate, Paste, and Translate. Translate is the one triggered by the original example in this issue.

The original Logisim did not have a redo menu item. This problem was introduced in LE at the time redo was introduced.

I checked the code in other subclasses of `Action` in other parts of the code and did not notice any that maintained a direct reference to a live collection of currently selected objects. They seem to hold that information (if it exists for that Action) in an separate collection of objects to process. They should not be bothered by new selection changes.

Hence, all the Actions that need to be corrected are contained in one file: `SelectionActions.java`.

I considered two basic approaches. 1) Change `doIt` so it doesn't depend on the live selection list and 2) Use `doIt` (largely as written) for the first invocation and a separate `redo` for subsequent invocations. Extracting the live selections from the various `doIt` methods did not look easy and would require modifications to several pieces of code.

On the other hand, I noticed that for all the operations except `Copy` and the `Copy` part of `Cut,` all the actual updates were built by `doIt` into a `CircuitMutation` object and that was used to do the action and its reverse was computed to be used in the `undo`. So I decided to try just hanging onto that original forward direction `CircuitMutation` as well as the reverse. That seems to work well.

I have implemented a solution that uses a subclass of `Action`, `SelectedComponentsAction`, that implements `doIt` as a call to `doItFirstTime` for the first call and `redo` for subsequent calls. It holds `xnForward` and `xnReverse`, the forward and reverse `CircuitMutations.` All the other Actions in the `SelectionActions.java` file are now subclasses of that class and their `doIt` method has been renamed `doItFirstTime`.

That leaves `Copy` and the Copy part of `Cut`. Those actually update the clipboard rather than the circuit. They were updating the clipboard on a `redo` using the currently selected objects rather than those selected at the time of the original Copy or Cut operation. They were also resetting the clipboard on an `undo` by holding the original clipboard and restoring it. I am not convinced that it is even necessary to undo a clipboard copy. It is possible that was introduced to "fix" the redo of the paste operation, but that is no longer necessary since the now properly fixed Paste operation no longer uses the clipboard to determine the objects to paste on an `redo`. Like the other operations, `redo` of `Paste` is now just a `xnForward` `CircuitMutation`. Nonetheless, It is possible something else is grabbing a clipboard value and needs this to remain. So I added to those operations the retention of the new clipboard so it can be reset on the redo without depending on a perhaps changed Selection.

Sorry for the length of this note, but I wanted to document what I am doing and why. A PR to fix this issue will follow.

maehne, 2024-09-01:
Thanks a lot @davidhutchens fir your detailed analysis and explanation of your approach how to fix it! It is very valuable!

davidhutchens, 2024-09-03:
Closing as completed with #2053.

closed 2024-09-03 by davidhutchens