logisim-evolution/logisim-evolution #1962: Java logo on subwindows

Shlok-Bhakta, 2024-02-07:
I'm not sure if this issue has been mentioned before but I haven't seen it, it seems that the sub windows where you can assign labels seems to have the default java logo. I figure I may as well mention it incase anyone wants something easy to fix 
[image]

maehne, 2025-06-01:
I can still reproduce the issue with Logisim-evolution built from 9af6af18 in a KDE Plasma session. To reproduce:

1. Create an empty circuit
2. Place a pin
3. Double-click pin to add a label

 The "Edit/specify components label" window has the wrong icon:

[image]

As other windows of Logisim-evolution's GUI get assigned the correct window icon, there seems to be some property not set for that specific small dialogue window.

shubd-dev, 2025-09-09:
Hi, Im unable to reproduce the issue 

[image]

wsl2 with ubuntu, openjdk-21

davidhutchens, 2025-09-10:
@maehne, I can't test this, but you have `Circuit circ` as a parameter. Does `circ.getProject().getFrame()` provide a suitable parent? At least the project frame seems to have the right icon from your picture. This may position the question frame at a different place so be sure that looks right. It seems to work on my Mac.

maehne, 2025-09-10:
Thanks, @davidhutchens, for this suggestion! It fixes the icon issue *and* the placement of the dialogue! It is now well centred above the current Logisim-evolution main window:

[image]

Before, it often opened on the wrong screen in multiple screen setups. I will submit a PR.