logisim-evolution/logisim-evolution #1966: Simulation Halted by Internal Error VHDL

Cosmic0921, 2024-02-18:
So I tried doing some VHDL to learn, on Logisim Evolution v3.8.0, and I got a error at the bottom saying "Simulation halted by internal error", and now the circuit won't run. I tried restarting the simulation, but I can't get it to work. Here's the code, it's an AND gate as a tutorial.
```
library ieee;
use ieee.std_logic_1164.all;
 
entity example_and is
  port (
    input_1    : in  std_logic;
    input_2    : in  std_logic;
    and_result : out std_logic
    );
end example_and;
 
architecture rtl of example_and is
  signal and_gate : std_logic;
begin
  and_gate   <= input_1 and input_2;
  and_result <= and_gate;
end rtl;
```
Also, unrelated issue, how do I install HDL? When I want to label things, it said "HDL Required", so if someone knows what to do, please LMK. Thanks!

primechaser, 2024-02-20:
did you activate "VHDL Simulation enabled" under menu item "Simulate"?

maehne, 2024-02-20:
Just enabling VHDL simulation is not sufficient. You need to install a supported VHDL simulator (currently only Mentor/Siemens' ModelSim or Questa Sim) and provide the path to the `bin/` directory with the simulator's executables `vlib`, `vcom`, `vsim`, etc. Be aware that co-simulation with a VHDL simulator is still experimental and quite fragile. I am not aware that anyone from the group of active developers is using this feature. So there are quite some limitations, you may run into once your VHDL examples get more complex.

Cosmic0921, 2024-02-25:
Het @maehne thank you for the response! I will use that then.

maehne, 2024-02-25:
> Edit: would intel questa be alright? The other "Siemens" is like business type.
Mentor Graphics was the original company behind ModelSim/Questa Sim, which was bought by Siemens several years ago. Altera/Intel licensed a  customised version of first ModelSim and later Questa Sim for distribution with their Quartus Prime FPGA design toolchain. They work well together with Logisim-evolution, too.

Cosmic0921, 2024-02-25:
Ok thank you!

closed 2024-02-25 by Cosmic0921