logisim-evolution/logisim-evolution #1968: Add a writable EEPROM with persistent data capability

MazinCE, 2024-02-22:
I would like to request the addition of a writable EEPROM element with the capability to store data persistently. It should allow the information to be readable even after the software is closed.

maehne, 2024-02-22:
PRs in this direction are welcome! However, first the requirements should be more precisely defined, and it should be checked whether they are not already (at least partially) fulfilled by the RAM or ROM component. The RAM component already has the attribute "Ram type" allowing to choose between "volatile" and "non volatile". It also offers in its context menu the possibility load/store an image. Maybe this already fulfils most of your needs?

MazinCE, 2024-02-22:
Hello maehne,

I usually work with RAM and haven't had much with ROMs. Would it be possible to include a write option for ROM elements?

MazinCE, 2024-02-22:
I want to use it as a storage device for a project I'm currently working on.

Quantodeluz, 2024-04-23:
> The RAM component already has the attribute "Ram type" allowing to choose between "volatile" and "non volatile". It also offers in its context menu the possibility load/store an image. Maybe this already fulfils most of your needs?

Actually, if you save data to a "Non-volatile" RAM, close the program, and then open it again, the data is not preserved.
I hope this helps to clarify why an EEPROM would be useful.

maehne, 2024-04-26:
@Quantodeluz: Thanks for suggesting a reasonable behaviour. Maybe the NVRAM content could be stored as part of the CIRC file so that it can be explicitly saved/discarded using the standard save dialog for Logisim schematics. PRs in this direction are welcome.

maehne, 2025-06-01:
@aguirre-lucas: Thanks for expressing your interest to contribute to Logisim-evolution! Our [Developer's corner](https://github.com/logisim-evolution/logisim-evolution/blob/main/docs/developers.md) provides some instructions to get started. In Logisim-evolution's online help (Help > User's Guide), you may find the chapters on “Value propagation” and “JAR libraries” helpful. Though, some information given there might not be any more up-to-date (help for updating the docs is always appreciated and if it is only by creating issues, where some information is not (any more) correct/up-to-date/missing).

Then, I would recommend having a look how the Logisim components in the Memory library have been implemented. The RAM and ROM models might be a suitable starting point for the EEPROM model. As stated [above](https://github.com/logisim-evolution/logisim-evolution/issues/1968#issuecomment-1959090592), I would suggest to first discuss/document the requirements/interface for the new EEPROM component (or enhancement to the existing RAM model), before actually implementing it. Once you start with implementing it in a new feature branch, you may create here a PR referencing this issue and mark it as draft to ask for feedback/support. Once you think the code is ready for thorough review, you can remove the draft attribute. We appreciate it if new component models are accompanied by documentation, tests, and a test circuit showcasing its functionality to ease our review.